OrcaSlicer/OrcaSlicer #8136: Bambu Lab LAN-only mode print doesn't start

all43, 2025-01-22:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0-dev

### Operating System (OS)

macOS

### OS Version

Sequoia 15.1

### Additional system information

Apple M3

### Printer

Bambu Lab A1 Mini, 

### How to reproduce

Load attached model (actually happening with almost any model recently), slice a plate (doesn't matter which or if you change anything or not), select a printer in LAN only model and click print.

### Actual results

Gettin network error:

Error code : -4020
Error desc: Failed to upload print file to FTP. Please
check the network status and try again.

[image]

### Expected results

Model prints as usual

### Project file & Debug log uploads

Model file here, but it happens with many downloaded models: [SAAB+99 (1).3mf.zip](https://github.com/user-attachments/files/18502299/SAAB%2B99.1.3mf.zip)

Log file for successful and not successful prints seems to look exactly the same, also noticed type in function name (desgin_id instead of design_id):

```
[warning]	2025-01-22 09:28:06.233174[Thread 0x00000001e61db840]:get_network_function, can not find function bambu_network_del_subscribe
[warning]	2025-01-22 09:28:06.235445[Thread 0x00000001e61db840]:get_network_function, can not find function bambu_network_get_model_id_from_desgin_id
[warning]	2025-01-22 09:28:06.235529[Thread 0x00000001e61db840]:get_network_function, can not find function bambu_network_get_profile_3mf
[error]	2025-01-22 09:28:06.909929[Thread 0x00000001e61db840]:3dx drivers module loading error: dlopen(/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient, 0x0005): tried: '/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient' (no such file), '/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient' (no such file)
[error]	2025-01-22 09:28:07.371322[Thread 0x00000001e61db840]:calc_exclude_triangles:Unable to create exclude triangles
[error]	2025-01-22 09:28:09.029027[Thread 0x00000001e61db840]:calc_exclude_triangles:Unable to create exclude triangles
[warning]	2025-01-22 09:28:09.515169[Thread 0x000000031e7d7000]:[Orca Updater]: online version=1.9.5.1, current_version=1.9.5.1, no need to download
[error]	2025-01-22 09:28:27.422614[Thread 0x00000001e61db840]:send_message_to_printer error: network_agent=0x600002ae3590, ret=-4, dev_id=0309DA462003092, json_str={"pushing":{"command":"pushall","push_target":1,"sequence_id":"20000","version":1}}, qos=0
[error]	2025-01-22 09:28:27.422870[Thread 0x00000001e61db840]:send_message_to_printer error: network_agent=0x600002ae3590, ret=-4, dev_id=0309DA462003092, json_str={"info":{"command":"get_version","sequence_id":"20001"}}, qos=1
```

### Checklist of files to include

- [x] Log file
- [x] Project file

### Anything else?

### Important context:
I've switched to LAN-only mode couple days ago, everything printed fine after the switch, but I had to select printer every time I start Orca Slicer and must be connected to the same wi-fi network otherwise printer discovery won't work. I think it is separate important issue - we should be able to add printer by IP and save it instead of using discovery every time.
If I resume my laptop from sleep with Orca Slicer already opened - printer disappears from the list and discovery doesn't work most of the time unless I restart Orca Slicer.

When printer is selected I could see live stream, select AMS filaments, adjust printer parts section, but sending a print usually fails at 30% with message as on screenshot.

If I create a new project from the scratch instead of downloading something - it prints fine.
Log however contains the same network-related warnings even for successful prints.

I thought issue might be related to the file size, but even if I slice small plate I'm getting the same error.

If export plate sliced file, I'm able to upload it using SFTP (I'm using FileZilla client) and print it by selecting on printer screen, so I assume ftp upload actually works.

I've tried removing (clicking on logout icon) and adding printer back - nothing has changed.

### Side note:
Probably unrelated, but I can't browser SD card, as it says feature not available in LAN-only mode, but I could browse it using SFTP client

dannyglover, 2025-01-23:
I'm having the same issue. Only happens in LAN mode. I can send prints to my A1 via FTPS with no issues, so it's not my network.

Thanks orca team.

arkku, 2025-01-23:
> we should be able to add printer by IP and save it instead of using discovery every time

Yes, please! I would like to use the X1Plus LAN mode with an access code without any Bambu Cloud involvement, but I can't get the printer discovered in that mode currently.

TheUnlimited64, 2025-01-26:
I also have this problem with my P1P in LAN Mode with V2.2.0

junelac, 2025-01-26:
I also have a similar problem with my X1C and OrcaSlicer 2.2 and 2.3dev on windows. This issue is not present on Bambu studio

Pedrat, 2025-01-29:
Same is happening here on my P1S, actually found a solution, which is to export the object (or objects) as STL, close the project, open a new one and import the STL. 

So, when printing with STL files it works, when doing it with .3mf they don't work. 
Sounds like might be an issue with the bambu network plugin itself? Would be nice to have this confimed.

barndawe, 2025-01-30:
Same issue here trying to print a 3MF I got from MakerWorld on my A1 (specifically [The Rock pencil holder](https://makerworld.com/en/models/558531#profileId-479040)). I can 'Send' it to the printer and then manually start it just fine but it fails to 'Print plate' directly from Orca with the error -4020 and FTP 9 as per the image below:
[image]

Saishy, 2025-01-30:
Same here, with this file: [Expandable cute shoe rack](https://makerworld.com/en/models/1025610#profileId-1018264)

vndk36, 2025-01-31:
Same Issue here and it's a bit anoying as I need to use the sd card to tranfer files to my printer. Also Calibration doesn't work in SD card mode

Pedrat, 2025-01-31:
Taking here a bit on the idea of @barndawe , here's also an example of something that didn't work, without exporting it as a STL first: https://makerworld.com/en/models/1029169#profileId-1011593

But here's something that did work out of the box, for some reason: https://makerworld.com/en/models/1021047#profileId-1002413

vndk36, 2025-01-31:
Well for my 3d models that I'm modeling myself, and exporting to step. Sometimes it work and sometimes it doesn't and just hangs there trying to upload.

I also tried to switch SD card without luck

deathlessdd, 2025-02-01:
This seems like an occering issue cause no one wants to use bambu cloud anymore I also have this issue. This was working before now it isnt? What is going on??

[image]

recrsn, 2025-02-01:
Bambu studio has an option to bind by IP address, useful for me when I access my devices over a VPN. Can't find the option in OrcaSlicer

samuelzamvil, 2025-02-02:
It's very odd to see this page [https://status.bambulab.com/](status.bambulab.com) when clicking on the error, because my printer is in LAN mode only and in Orca I have stealth mode enabled. So I shouldn't need bambu services at all.

cryptonym64, 2025-02-03:
> Maybe Bambulabs is injecting some code into the 3mf files which forces connection to their servers?

I suspect so, as their rewards system awards based on # of prints and such. They likely are tracking us printing all along.

goldsoundz80, 2025-02-03:
> Maybe Bambulabs is injecting some code into the 3mf files which forces connection to their servers?

This is what I thought too, but some 3MF files from MakerWorld work, and some don't. I thought it might be something to do with whether the 3MF file was created in Bambu Studio or Orca, but I've tried it on one of my own models that I know I created in Bambu Studio, and it prints with no error from Orca.

The Yachty profile from here doesn't work:
https://makerworld.com/en/models/1004947#profileId-983581

One of my models (created in Bambu Studio fairly recently) which does work:
https://makerworld.com/en/models/854563#profileId-803546

Nikakastoney, 2025-02-04:
I have encountered this error several times as well, and I believe the suspicion about the 3mf files might be correct. I opened one of the files in Orca Slicer and exported it as an STL. Then I opened the STL File as a new project, and after that, it was able to send the print job to my A1 mini. [https://makerworld.com/de/models/1016252?from=search#profileId-996550](url)

samuelzamvil, 2025-02-05:
I found a workaround.
If you're experiencing this issue; open a new project, then import the 3mf, it imports the settings and geometry. Then try slicing and printing over LAN.

goldsoundz80, 2025-02-05:
> If you're experiencing this issue; open a new project, then import the 3mf, it imports the settings and geometry.

This doesn't work for me, when I import the 3MF into Orca (rather than open it), the geometry is imported but not the settings.

E.g., I tried it with the Yachty one:
https://makerworld.com/en/models/1004947#profileId-983581

**Open**ing the 3MF brings with it settings such as using 3 wall loops. **Import**ing the 3MF keeps the default settings (or maybe what I used last).

samuelzamvil, 2025-02-05:
> This doesn't work for me, when I import the 3MF into Orca (rather than open it), the geometry is imported but not the settings.

Sorry for the false hope. This is a misunderstanding on my part.

I believe I know why I got it to work. I saved the print settings as a custom preset and when I imported the models from the 3mf than it kept the setting. I'm speculating but I think this means it won't work for 3mf files with separate settings for individual modules.

goldsoundz80, 2025-02-05:
That’s interesting because the Yachty one does have two separate bits, it slows down for the flag, which is a separate body…

igiannakas, 2025-02-14:
@Noisyfox maybe one to take a look at? 🤔

Noisyfox, 2025-02-14:
Ok I found the issue. Will raise a PR soon.

samuelzamvil, 2025-02-14:
Not nefarious at all. Thanks for coming up with a fix.

closed 2025-02-15 by Noisyfox

vndk36, 2025-02-20:
Coming back to this thread, the fix looks like it it working now from my small test. thx @Noisyfox  for the fix!

PR #8407 (2025-02-15): FIX:filter the characters of model names in the model mall
If the model comes from model mall, the name from the mall will be used when sending and printing. When there are special characters in the name, it will cause the sending to fail.

cherry picked from commit https://github.com/bambulab/BambuStudio/commit/1bcf30c39c648763952703c9060573baa2782f87

Thanks BambuLab!

Fix #8136