OrcaSlicer/OrcaSlicer #8137: Feature Request: Ability to change line print order to address benchy hull line via modifier.

GCheung55, 2025-01-22:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

When printing bins, boxes, vases, benchies, and the like, the benchy hull line issue appears due to the "sudden transition from sparse infill into full top layers", as described by [Prusa devs investigation](https://help.prusa3d.com/article/the-benchy-hull-line_124745).

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

The solution I'd like is the ability to change the line print order via modifier, height-range and regular modifier. The UI could be drag/drop ordered list, or drop down menus for each type of line.

[Prusa devs modified the line print order by hand and saw improvements.](https://help.prusa3d.com/article/the-benchy-hull-line_124745#:~:text=Then%20there%E2%80%99s%20the%20order%20of%20elements%20inside%20a%20layer.%20For%20some%20reason%2C%20we%20got%20better%20results%20when%20always%20printing%20the%20deck%20perimeters%20first%2C%20then%20infill%20of%20the%20deck%20and%20then%20the%20rest%20of%20the%20layer.%20We%20manually%20edited%20this%20order%20using%20a%20text%20editor.).

> Then there’s the order of elements inside a layer. For some reason, we got better results when always printing the deck perimeters first, then infill of the deck and then the rest of the layer. We manually edited this order using a text editor.

### Describe alternatives you've considered

1. I've tried using multiple modifier where one had walls but surfaces or infill, and another only had walls, but I could not get the print order to change.
2. I've tried adding a void inside the model to prevent the benchy hull line, but it still appears.
3. I've tried printing outer walls slower, but the benchy hull line still appears.

### Additional context

Discussion: [Investigation to prevent benchy hull line while printing Gridfinity bins generated from gridfinity-rebuilt-openscad](https://github.com/kennetek/gridfinity-rebuilt-openscad/discussions/244)

GCheung55, 2025-01-26:
The user going by Divide in the discord pointed me to look at https://github.com/SoftFever/OrcaSlicer/commit/a53f12489e7c1c0a6c5a64302a70463343dc4c2b.

I THINK I should be looking for how `ExtrusionPath`, defined in `src/libslic3r/ExtrusionEntity.hpp`, is being used because it's has a role. And if I can figure out a place to change the order of extrusion paths (what I had called `line`).

sleewok, 2025-01-28:
@rbq this is a really good idea. It could potentially help with curled edges as well.

GCheung55, 2025-01-30:
@igiannakas Since you were experimenting with fixes to the Benchy Hull Line, may I get your thoughts on this feature request?

GCheung55, 2025-02-03:
@SoftFever @Noisyfox Love to get your thoughts too.

igiannakas, 2025-02-03:
Help me understand what you’re thinking if you don’t mind.

Basically you’re printing inner outer and want to convert to inner outer inner/outer inner  where the solid layer happens? Or is it something different?

GCheung55, 2025-02-03:
@igiannakas It's something different.

Current print order is:
1. Perimiter/Infill (or Infill/Perimiter).
2. Internal bridge, solid infill, or top surfaces.

The feature request is to change the print order to:
1. Internal bridge, solid infill, or top surfaces.
2. Perimiter/Infill (or Infill/Perimiter).

The rationale is based on what the Prusa team described as one of their solutions to the hull benchy line, where they manually changed the gcode. From the Prusa dev blog:

> Then there’s the order of elements inside a layer. For some reason, we got better results when always printing the deck perimeters first, then infill of the deck and then the rest of the layer. We manually edited this order using a text editor.

My understanding here is that they printed the deck permiters and infill, and then the hull permiters and infill. This is what I'd like to have the ability to do. I recognize it's slightly different from my request of controlling the order, since what the blog describes is changing not only the order, but also as if by part. I was hoping that requesting the line print order might be an easier lift than to change the part order - happy to be wrong though.

GCheung55, 2025-02-03:
@igiannakas I see that internal bridging, solid infill, top surfaces, AND sparse infill will print before the perimeters when `Print Infill First` is selected. 

However, it isn't always the case. Case in point, the model I'm using has a negative part that creates a gap between the internal bridge / solid infill / top surface from the outer walls of a bin. 

I've attached the test model with the negative part (void.stl), and enabled `Print Infill First`. Apologies for the zip file, Github doens't allow 3mf files to be attached. At line 31, you can see that the print order goes:

1. Bin - Sparse Infill
2. Bin - Outerwall
3. Bin - Innerwall
4. Void (Outerside, closest to the bin wall) - Outerwall
5. Void (Outerside) - Innerwall
6. Bin - Internal Bridge
7. Void - (Innerside, furthest from the bin wall) - Outerwall
8. Void - (Innerside) - Innerwall

According to what you're saying, the internal bridging should have gone first but doesn't.

[void-test.3mf.zip](https://github.com/user-attachments/files/18647576/void-test.3mf.zip)

GCheung55, 2025-02-07:
I was pointed to https://github.com/SoftFever/OrcaSlicer/blob/main/src/libslic3r/ExtrusionEntity.hpp, but I haven't figured out where instances are being created and where the line print order can be changed.

GCheung55, 2025-04-10:
@igiannakas I also tried saving the STL file with the void in place from Orca and the same order occurs. Bug?