OrcaSlicer/OrcaSlicer #8139: Move Gizmo not working as expected

discip, 2025-01-22:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

latest dev build

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

-

### How to reproduce

[image]

### Actual results

-

### Expected results

-

### Project file & Debug log uploads

-

### Checklist of files to include

- [ ] Log file
- [ ] Project file

discip, 2025-01-22:
Edit:
Since I used a sphere as the 2nd object, I assumed that **`Lay on Face`** was also affected, but that is actually not the case. 🤦‍♂️

discip, 2025-01-22:
Since I used a sphere as the 2nd object, I assumed that **`Lay on Face`** was also affected, but that is actually not the case. 🤦‍♂️

Noisyfox, 2025-01-22:
It was actually caused by the lay on face gizmo. Already found the solution, will push a fix soon.

PR #8141 (2025-01-22): Fix issue that gizmo grabbers stop working
Fix #8139
Follow up of a66fc86e203572f84a0761d607fc7e0170e3afd7
closed 2025-01-22 by Noisyfox