OrcaSlicer/OrcaSlicer #8144: Non-Bambu/Generic/eSun/etc. filaments such as Elegoo not available to choose

ksignorini, 2025-01-23:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

macOS

### OS Version

Sequoia 15.2

### Printer

Bambu A1

### How to reproduce

1. Click "Set filaments to use" button.
2. Choose All filament types, All vendors, and All presets. Then Confirm.
3. Click the dropdown on a filament number and look for a preset not available in regular Bambu Studio, like Elegoo.

### Actual results

See that the only ones listed are Bambu, Generic, and eSun. These match the only options that are available in Bambu Studio. None of the Orca Slicer-only presets like Elegoo are available to choose.

### Expected results

Elegoo (and other) presets should be available as choices for filaments.

### Project file & Debug log uploads

No project file required. Even happens on band new, empty project.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

[image]

[image]

Burbsi, 2025-01-31:
Same issue on windows10, chose generic PCTG and it does not show up. The troubleshoot page suggests editing a JSON file, i don't have any files in the named folder.

Beh3m0th, 2025-07-23:
Same issue here using v2.3.0 on Ubuntu 22.04