OrcaSlicer/OrcaSlicer #8145: Support proprietary nvidia drivers on wayland linux sessions

powermad80, 2025-01-23:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0-dev

### Operating System (OS)

Linux

### OS Version

Fedora 41 KDE

### Additional system information

CPU: Ryzen 3700X
GPU: Nvidia RTX 3080
GPU Driver Version: 565.77

### Printer

N/A

### How to reproduce

1. Launch any version of Orca Slicer for Linux during a wayland desktop session running the latest NVidia proprietary drivers
2. Notice that the entire build plate preview is completely missing
3. Close Orca Slicer
4. Set an environment variable in flatseal or wherever else appropriate reading `__EGL_VENDOR_LIBRARY_FILENAMES=/usr/lib/x86_64-linux-gnu/GL/default/share/glvnd/egl_vendor.d/50_mesa.json`
5. Relaunch Orca Slicer

### Actual results

The build plate preview and all its controls are now visible, but performance is extremely diminished, and that all actions from manipulating items on the build plate preview to even resizing the window are incredibly choppy.

### Expected results

Step 4 in the reproduction steps should not be required to run the slicer with full functionality, and NVidia drivers should be able to be utilized to run the slicer.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [x] Log file
- [x] Project file

lkarlslund, 2025-01-30:
I have this problem on Ubuntu 25.04 Wayland with Nvidia

williamccrobinson, 2025-01-31:
exactly same problem here on the latest v2.2.0 release. 
Using arch linux + gnome 47.3 + wayland

### OrcaSlicer Version
2.2.0

### Operating System (OS)
Linux

### OS Version
Arch Linux x86_64
Kernel: 6.12.10-arch1-1

### Additional system information
CPU: Ryzen Threadripper 1950X
GPU: Nvidia GTX 1080 Ti
GPU Driver Version: 565.77

Noisyfox, 2025-02-09:
Could you try the build of this PR https://github.com/SoftFever/OrcaSlicer/pull/6621

powermad80, 2025-02-09:
> Could you try the build of this PR [#6621](https://github.com/SoftFever/OrcaSlicer/pull/6621)

Installed the flatpak from the build, unfortunately no change on my end.

Fedora 41 KDE, Ryzen R3700X, RTX 3080 (driver 565.77)

powermad80, 2025-02-09:
To add some more detail, when running the flatpak through terminal the output is entirely this on startup:

[image]

Noisyfox, 2025-02-11:
[image]

Weird... I couldn't reproduce this issue. It works just fine with my Fedora installation. Did I miss anything? Never use Linux on a bare machine before.

KaiMartin, 2025-03-03:
On my Debian machine (trixie with nvidia proriatary drivers) the slice preview of moderately sized models renders very slowly. Just rotating the model becomes a 0.5 FPS waiting game. I notice high processor load. Seems like all rendering is done in software and the graphics card is sitting idle.
On the command line I get the same errors as in the screenshot of powermad80 - lots of GTK errors and somewhere in between lines about failures to create dri2 screens. If I skip the env variables, the number of egl failure lines reduces to two. But the previews are still sluggish.

I was able to get decent performance from the same hardware after I switched to the X11 version of plasma 6. There were a few glitches in the UI like inappropriately sized fonts on buttons. Probably related to me using a hidpi monitor. The rest of my plasma desktop felt a little weird, too. But working with the actual 3D model is so much better that I will log out of plasma-wayland and log into X11 to work with orca until this issue is fixed.

BTW, freecad has issues with wayland, too. As a workaround, the environment variable XDG_SESSION_TYPE=X  makes freecad not use X11. Unfortunately, thins env does not seem to influence orca. Is there any other way to make orca use X11 short of logging in a full fledged X11 session?

Sheenomatics, 2025-03-16:
> __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json WEBKIT_FORCE_COMPOSITING_MODE=1 WEBKIT_DISABLE_COMPOSITING_MODE=1 WEBKIT_DISABLE_DMABUF_RENDERER=1

setting these environment variables solved it too for me with Nvidia on each Wayland Distro (Tested on KDE Plasma, Cosmic Desktop (new PopOS), Ubuntu 24.04). Please solve this inside the Appimage.

h4lm3th34d, 2025-04-30:
Setting those ENV variables crashed OrcaSlicer (flatpak) for me in Fedora42

Rihan9, 2025-07-06:
Hi:
OS: EndeavourOS (arch based)
OpenGL version: OpenGL ES 3.2 
Nvidia Version: NVIDIA 575.64
orca-slicer: 2.3.1-dev

Same problem, no builplate loaded.
## Workaround:
- Install wine-ge or lutris for an easy configuration
- Download the portable version and run once from wine-ge/lutris
- The program will not load the welcome page or the first time flow, but it will create the folder structure.
- Overwrite all the content in `/home/{{USERNAME}}/.wine/drive_c/users/{{USERNAME}}/AppData/Roaming/OrcaSlicer/` with the content of a working OrcaSlicer, or the content of the flatpak version where you have setted-up your printer and your profile
- Rerun Orca from lutris/wine. Now you have a sketchy, working environment

[image]

pm0u, 2025-08-12:
Attempting to offload Orca to my eGPU - on manjaro linux using the version from the AUR @ModMonster your fixes did not seem to work - I get rendering (an improvement) but then everything freezes and the app crashes. I did have to tweak the AppRun a bit as the directories seemed to be off. Then run `prime-run /opt/orcaslicer/bin/AppRun` and I get the crash.

Rotziboisseur, 2025-08-15:
Hi, just bumping this issue – still happening on my Arch Linux setup (KDE Plasma Wayland, Nvidia RTX 5070, driver 555.58.02).
**Context & testing details:**
- Tried Flatpak, AUR build, and AppImage (with --appimage-extract-and-run) → viewport remains grey/empty.
- Already used common workarounds (GDK_BACKEND=x11, __GLX_VENDOR_LIBRARY_NAME=nvidia, etc.) — none resolved the issue.
- Reset config (~/.config/OrcaSlicer) didn’t help.
- My system: XDG_SESSION_TYPE=wayland, OpenGL renderer is confirmed as Nvidia (via glxinfo output).
- Notably, the only "fix" that worked partially was forcing Zink (OpenGL over Vulkan via Mesa) as seen in some forums.

This bug has been lingering for a certain time — adding here to strengthen the case for a fix.

Let me know if I can provide logs or configurations to help troubleshoot further.

BRUrban, 2025-08-16:
This doesn't resolve the issue and doesn't enable Wayland support, but since I am also experiencing this issue and none of the environment variable workarounds using EGL, MESA, and WebKit are solving it for me, I wanted to note that the approach from a similar but distinct issue regarding specifically the 3D view failing after nvidia update is working for me with the 2.3.0 AppImage:

https://github.com/SoftFever/OrcaSlicer/issues/6433#issuecomment-3037295054
(Set `XDG_SESSION_TYPE=x11` and `GBM_BACKEND=dri` for it)

I would rather have Wayland playing nicely, but this at least allowed usage of the app.

alemdq, 2025-08-17:
I made OrcaSlicer work by running it inside a nested X11 server, because my system runs Wayland with KDE and the AppImage wouldn’t launch properly.
First, I started Xephyr on display :2 while telling it to use my host display :1.
Then I launched a minimal window manager, openbox, inside that Xephyr session.
Finally, I set DISPLAY=:2 with GDK_BACKEND=x11 and QT_QPA_PLATFORM=xcb before running the OrcaSlicer AppImage, and it opened correctly inside the Xephyr window. 

I’m currently trying to do the same with Creality print buy or just won’t work 🥲

pm0u, 2025-08-20:
I was actually able to get the Flatpak working just out of the box, using Nouveau drivers currently.

using `WEBKIT_DISABLE_COMPOSITING_MODE=1` in flatseal gets rid of some weird artifacts on the welcome screen.. but otherwise it seems to be good to go.

ZeroBomb, 2025-09-13:
https://github.com/SoftFever/OrcaSlicer/issues/8145#issuecomment-3193725130

This worked for me on fedora 42 running nvidia driver version 580.82.07

i-hayes, 2025-11-18:
I am using Opensuse Tumbleweed with Nvidia 3050 video card.

When I opened OrcaSlicer 2.3.1 today the preview window was again blank. Orca slicer is installed as a flatpak. 

My previous post listed the fixed for this issue that worked for me. Now none of the above workarounds provide a fix.

The appimage version 2.3.1 still works. 

I tried the nightly 2.3.2 still no preview.

andree-rehnberg, 2026-02-05:
Experiencing the same issue on Fedora 43 on a machine with an RTX 3080. Setting `GBM_BACKEND=dri` in flatseal does work but makes the preview extremely sluggish. Tested  the settings in https://github.com/OrcaSlicer/OrcaSlicer/issues/8145#issuecomment-2623878509 but it made orca crash. Would love to see this solved.

Lucretia, 2026-02-27:
Ok, what's the magic incantation to get this application to even start without segfaulting these days?

I was using: `env -u WAYLAND_DISPLAY XDG_SESSION_TYPE=x11 WEBKIT_FORCE_COMPOSITING_MODE=1 WEBKIT_DISABLE_COMPOSITING_MODE=1 GBM_BACKEND=dri` followed by the appimage, but that and the installed binary on gentoo both crash on startup.

```bash
$ env -u WAYLAND_DISPLAY XDG_SESSION_TYPE=x11 WEBKIT_FORCE_COMPOSITING_MODE=1 WEBKIT_DISABLE_COMPOSITING_MODE=1 GBM_BACKEND=dri  orca-slicer 
[2026-02-27 21:20:24.057232] [0x00007fe246fa5380] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1

(orca-slicer:16913): Gtk-CRITICAL **: 21:20:24.291: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:16913): Gtk-CRITICAL **: 21:20:24.292: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:16913): Gtk-CRITICAL **: 21:20:24.293: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:16913): Gtk-CRITICAL **: 21:20:24.294: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:16913): Gtk-CRITICAL **: 21:20:24.295: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:16913): Gtk-CRITICAL **: 21:20:24.295: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:16913): Gtk-CRITICAL **: 21:20:24.295: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:16913): Gtk-CRITICAL **: 21:20:24.296: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:16913): Gtk-CRITICAL **: 21:20:24.296: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:16913): Gtk-CRITICAL **: 21:20:24.296: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
pci id for fd 18: 10de:2482, driver (null)
pci id for fd 28: 10de:2482, driver (null)
kmsro: driver missing
libEGL warning: pci id for fd 18: 10de:2482, driver (null)

pci id for fd 28: 10de:2482, driver (null)
kmsro: driver missing
libEGL warning: egl: failed to create dri2 screen
libEGL warning: egl: failed to create dri2 screen

(orca-slicer:16913): Gtk-CRITICAL **: 21:20:24.470: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:16913): Gtk-CRITICAL **: 21:20:24.471: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:16913): Gtk-CRITICAL **: 21:20:24.471: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:16913): Gtk-CRITICAL **: 21:20:24.471: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:16913): Gtk-CRITICAL **: 21:20:24.471: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:16913): Gtk-CRITICAL **: 21:20:24.471: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:16913): Gtk-CRITICAL **: 21:20:24.471: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:16913): Gtk-CRITICAL **: 21:20:24.471: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:16913): Gtk-CRITICAL **: 21:20:24.471: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:16913): Gtk-CRITICAL **: 21:20:24.475: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:16913): Gtk-CRITICAL **: 21:20:24.475: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:16913): Gtk-CRITICAL **: 21:20:24.475: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:16913): Gtk-CRITICAL **: 21:20:24.475: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:16913): Gtk-CRITICAL **: 21:20:24.475: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:16913): Gtk-CRITICAL **: 21:20:24.475: gtk_widget_set_size_request: assertion 'width >= -1' failed
[... 340 lines cut ...]
     Description:         NVIDIA wayland EGL external platform library

[I] gui-libs/egl-wayland2
     Available versions:  (~)1.0.0_rc 1.0.1_pre20260109 {ABI_MIPS="n32 n64 o32" ABI_S390="32 64" ABI_X86="32 64 x32"}
     Installed versions:  1.0.1_pre20260109(20:10:30 03/02/26)(ABI_MIPS="-n32 -n64 -o32" ABI_S390="-32 -64" ABI_X86="64 -32 -x32")
     Homepage:            https://github.com/NVIDIA/egl-wayland2/
     Description:         NVIDIA wayland EGL external platform library, version 2

[I] kde-plasma/kwayland
     Available versions:  
     (5)    [M]5.116.0-r1(5/5.116)^t
     (6)    6.5.5^t ~6.6.1^t
       {debug doc test}
     Installed versions:  6.5.5(6)^t(15:18:50 18/02/26)(-debug -test)
     Homepage:            https://invent.kde.org/frameworks/kwayland
     Description:         Qt-style API to interact with the wayland-client API

[I] x11-base/xwayland
     Available versions:  24.1.9^t **9999*l^t {libei selinux systemd test unwind xcsecurity}
     Installed versions:  24.1.9^t(23:51:58 08/11/25)(libei -selinux -systemd -test -unwind -xcsecurity)
     Homepage:            https://wayland.freedesktop.org/xserver.html
     Description:         Standalone X server running under Wayland

Found 10 matches
```