OrcaSlicer/OrcaSlicer #8146: crash when slicing

Lljjcc625, 2025-01-23:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.2.0 and 2.3.0 dev

### Operating System (OS)

Windows

### OS Version

win10 22h2

### Additional system information

12th Gen Intel(R) Core(TM) i5-12490F   3.00 GHz
32.0 GB
rtx 4070ti

### Printer

all printer

### How to reproduce

1.Open orcaslider
2.import a 3mf/stl/step
3.click slice plate
1.打开 orca
2.随便导入什么
3.开始切片

### Actual results

the orcaslicer will crash
orca崩溃了
https://github.com/user-attachments/assets/09236f04-d41d-4879-af67-db5d1ff02abc

### Expected results

just crash

### Project file & Debug log uploads

project :i try anything project,but the problem is same
log file:
[warning]	2025-01-23 10:26:00.100916[Thread 0x00001f1c]:Failed to load to hints.cereal. File does not exists. C:\Users\LIDA\AppData\Roaming\OrcaSlicer\user\hints.cereal
[warning]	2025-01-23 10:26:00.144424[Thread 0x00001f1c]:Slic3r::NetworkAgent::get_network_function, can not find function bambu_network_get_model_id_from_desgin_id
[warning]	2025-01-23 10:26:00.144424[Thread 0x00001f1c]:Slic3r::NetworkAgent::get_network_function, can not find function bambu_network_get_profile_3mf
[warning]	2025-01-23 10:26:04.778754[Thread 0x00002738]:[Orca Updater]: online version=1.9.5.1, current_version=1.9.5.1, no need to download
[error]	2025-01-23 10:29:33.746206[Thread 0x00001f1c]:Slic3r::GUI::GuideFrame::LoadProfileFamily: parse C:\Program Files\OrcaSlicer\resources\profiles\Creality.json got exception: [json.exception.type_error.302] type must be string, but is null
[error]	2025-01-23 10:29:33.773713[Thread 0x00001f1c]:Slic3r::GUI::GuideFrame::LoadProfileFamily: parse C:\Program Files\OrcaSlicer\resources\profiles\Elegoo.json got exception: [json.exception.type_error.302] type must be string, but is null
[error]	2025-01-23 10:29:33.775714[Thread 0x00001f1c]:Slic3r::GUI::GuideFrame::LoadProfileFamily: parse C:\Program Files\OrcaSlicer\resources\profiles\Eryone.json got exception: [json.exception.type_error.302] type must be string, but is null
[error]	2025-01-23 10:29:33.858713[Thread 0x00001f1c]:Slic3r::GUI::GuideFrame::LoadProfileFamily: parse C:\Program Files\OrcaSlicer\resources\profiles\Kingroon.json got exception: [json.exception.type_error.302] type must be string, but is null

### Checklist of files to include

- [x] Log file
- [x] Project file

Lljjcc625, 2025-01-24:
today it is output a new log
[warning]	2025-01-24 08:57:21.302328[Thread 0x000048b0]:Slic3r::NetworkAgent::get_network_function, can not find function bambu_network_get_model_id_from_desgin_id
[warning]	2025-01-24 08:57:21.303327[Thread 0x000048b0]:Slic3r::NetworkAgent::get_network_function, can not find function bambu_network_get_profile_3mf
[error]	2025-01-24 08:57:24.502048[Thread 0x000048b0]:Slic3r::NetworkAgent::connect_server error: network_agent=00000266FCD894E0, ret=-2
[error]	2025-01-24 08:57:25.503049[Thread 0x000013b4]:Error getting: `check_new_version_sf`: HTTP 0, curl:Timeout was reached:
Connection timed out after 1001 milliseconds
[Error 28]
[warning]	2025-01-24 08:57:26.893584[Thread 0x000013c4]:[Orca Updater]: online version=1.9.5.1, current_version=1.9.5.1, no need to download

dinoelT, 2025-02-09:
I have very similar logs on Windows, but without the crash. It just can't find my A1 mini printer in LAN mode:

> [warning]	2025-02-08 12:04:43.188898[Thread 0x00008bbc]:Slic3r::NetworkAgent::get_network_function, can not find function bambu_network_get_model_id_from_desgin_id
> [warning]	2025-02-08 12:04:43.189398[Thread 0x00008bbc]:Slic3r::NetworkAgent::get_network_function, can not find function bambu_network_get_profile_3mf
> [error]	2025-02-08 12:04:47.920591[Thread 0x00008bbc]:Slic3r::GUI::PartPlate::calc_exclude_triangles:Unable to create exclude triangles
> [error]	2025-02-08 12:04:47.926129[Thread 0x00005dfc]:Unable to open the fileC:\Users\LeonidTanas\Downloads\hexasphericons.3mf
> [error]	2025-02-08 12:04:47.926629[Thread 0x00008bbc]:Unable to open the fileC:\Users\LeonidTanas\Downloads\arduino-2wd-robot-chassis-1.snapshot.3\frame.stl
> [warning]	2025-02-08 12:04:50.013302[Thread 0x00008bbc]:12:04:50 PM: Navigation request to 'file:///C:/Program%20Files/OrcaSlicer/resources/web/homepage/index.html?lang=en_US' (target='')
> [error]	2025-02-08 12:04:52.830371[Thread 0x00008bbc]:Slic3r::GUI::PartPlate::calc_exclude_triangles:Unable to create exclude triangles
> [warning]	2025-02-08 12:04:53.019414[Thread 0x00008bbc]:12:04:53 PM: Document loaded; url='file:///C:/Program%20Files/OrcaSlicer/resources/web/homepage/index.html?lang=en_US'
> [warning]	2025-02-08 12:04:53.120290[Thread 0x00008bbc]:12:04:53 PM: Script message received; value = {"sequence_id":1739012693,"command":"get_login_info"}, handler = wx
> [warning]	2025-02-08 12:04:53.121288[Thread 0x00008bbc]:12:04:53 PM: Script message received; value = {"sequence_id":1739012693,"command":"get_recent_projects"}, handler = wx
> [warning]	2025-02-08 12:04:53.132304[Thread 0x00008bbc]:12:04:53 PM: Script message received; value = {"sequence_id":1739012693,"command":"modelmall_model_advise_get"}, handler = wx
> [warning]	2025-02-08 12:04:53.134322[Thread 0x00008bbc]:12:04:53 PM: Navigation complete; url='file:///C:/Program%20Files/OrcaSlicer/resources/web/homepage/index.html?lang=en_US'
> [warning]	2025-02-08 12:04:54.605105[Thread 0x00006db4]:[Orca Updater]: online version=1.9.5.1, current_version=1.9.5.1, no need to download
> [warning]	2025-02-08 12:05:02.340823[Thread 0x00008bbc]:12:05:02 PM: Script message received; value = {"sequence_id":1739012702,"command":"homepage_newproject"}, handler = wx
> [error]	2025-02-08 12:05:02.388284[Thread 0x00008bbc]:Slic3r::GUI::PartPlate::calc_exclude_triangles:Unable to create exclude triangles
> [error]	2025-02-08 12:06:16.080793[Thread 0x00008bbc]:Slic3r::GUI::GuideFrame::LoadProfileFamily: parse C:\Program Files\OrcaSlicer\resources\profiles\Creality.json got exception: [json.exception.type_error.302] type must be string, but is null
> [error]	2025-02-08 12:06:16.126826[Thread 0x00008bbc]:Slic3r::GUI::GuideFrame::LoadProfileFamily: parse C:\Program Files\OrcaSlicer\resources\profiles\Elegoo.json got exception: [json.exception.type_error.302] type must be string, but is null
> [error]	2025-02-08 12:06:16.284974[Thread 0x00008bbc]:Slic3r::GUI::GuideFrame::LoadProfileFamily: parse C:\Program Files\OrcaSlicer\resources\profiles\Kingroon.json got exception: [json.exception.type_error.302] type must be string, but is null
> [error]	2025-02-08 12:07:57.520917[Thread 0x00008bbc]:Slic3r::GUI::PartPlate::calc_exclude_triangles:Unable to create exclude triangles

closed 2025-05-18 by github-actions[bot]