OrcaSlicer/OrcaSlicer #8151: Prime tower size doesn't response to setting. Minimum size too BIG / huge for multiextruders

LehoooJ, 2025-01-23:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

Dev 2.3.0 1b1288c

### Operating System (OS)

Windows

### OS Version

Win11

### Printer

Custom QIDI iFast

### How to reproduce

a) Set Prime tower volume to any nummber
b) set Minimum purge on wipe

### Actual results

a) no change if 1 or 500
b) 1st extruder minumum about 40, 2nd minimum 150 (even I set 1mm3)

### Expected results

Smaller prime tower, this works better Prusa.... but i like Orca more for many reasons...

### Project file & Debug log uploads

[CylinderMC.zip](https://github.com/user-attachments/files/18517672/CylinderMC.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

### Anything else?

[image]

[image]

igiannakas, 2025-01-23:
Have you set it up like so? 

[image]

igiannakas, 2025-01-23:
> HEy, nice! Thank you! I was stuck on setting Purge ON, till I set Single,,,,unchecked Purge OFF and than slelected Mulit extruder again.
> 
> [image]

Perfect! Glad it worked out;)

cdvankammen, 2025-03-15:
I have been finding that this works with a plate that has one model but the moment that you put multiple models and or up to filling the bed ... the purge tower becomes huge and unable to change the direction. I am using the KOBRA 3 print profile default as the base... which removes the filament purge amount stats as well. 

Any thoughts on how we can work to fix this or I can help code it up?

GreavesDesignerOne, 2025-03-15:
I was having the same issue for quite a while in version 2.2, until I found this thread. Thank you LehoooJ for reporting this! The solution is not obvious.

I'm glad that this is actually functional. It seems the minimum prime tower prime volume (in version 2.2 anyway) is around 10mm^3 (the model does not appear to change below that setting), and that's still a bit too large for the IDEX process and small parts that I'm working on, though I can live with it. There does seem to be a UI issue here however.

cdvankammen, 2025-03-15:
So I guess I'm confused I have tried this, usually the issues arise when I have multiple colored parts on the build plate at the same time. And I've tried to click and unclick The Purge however it doesn't seem to fix itself. If there is a open source GitHub for any cubic I think that they have theirs working without major issues so I can take a look at that code and see what might be causing the bug.

closed 2025-06-21 by github-actions[bot]