OrcaSlicer/OrcaSlicer #8152: Cannot type the letter `c` as it is colliding with keybinding

snadorp, 2025-01-23:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

macOS

### OS Version

15.2 (24C101)

### Printer

Bambulab A1 Mini

### How to reproduce

In the Preview Tab
Go to: Post-Processing Scripts
Type anything containing a c like:
`/home/user/scripts/cccccc/main.py`

### Actual results

The script field will show and contain the following string:
`/home/users/sripts//main.py`

### Expected results

The actual typed code:
`/home/user/scripts/cccccc/main.py`

### Project file & Debug log uploads

no need

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

The keybinding of `Show G-code Window` is triggering when typing.

[image]

There are two solutions to this:
1. Change the keybinding to make use of a modifier key.
2.  Ignore key bindings in the script editor field. (Added benefit: Undo (STRG-Z) would be working in the field)

Noisyfox, 2025-02-09:
Seems to be a macOS specific issue.

ShantanuNair, 2025-03-17:
Facing this as well, when trying to add a Post-processing Script.

PR #9136 (2025-04-05): Fix shortcut collision on macOS
 Fix #8152
commit 85d46d29 (2025-04-05): Fix shortcut collision on macOS (#9136)

 Fix #8152
closed 2025-04-05 by SoftFever