OrcaSlicer/OrcaSlicer #8155: CLI load-settings - unknown config type of file

debauer, 2025-01-23:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Linux

### OS Version

Archlinux

### Printer

Bambulabs A1, Ender3 V3 KE, Ender3 Pro, Ender3

### How to reproduce

run 
```
orca-slicer --debug 5 --export-slicedata --outputdir="." --slice=0 --load-filaments=filemant.json --load-settings="process.json;machine.json" "bin_z8_x3_y4.stl"
```
[stl.zip](https://github.com/user-attachments/files/18519479/stl.zip)
[process.json](https://github.com/user-attachments/files/18519419/process.json)
[machine.json](https://github.com/user-attachments/files/18519418/machine.json)
[filament.json](https://github.com/user-attachments/files/18519420/filament.json)

### Actual results

```
[2025-01-23 12:00:16.768920] [0x00007b63558efc80] [trace]   Initializing StaticPrintConfigs
[2025-01-23 12:00:16.778064] [0x00007b63558efc80] [warning] cli mode, Current OrcaSlicer Version 01.09.05.51
[2025-01-23 12:00:16.778101] [0x00007b63558efc80] [info]    Will start to read model file now, file count :1

[2025-01-23 12:00:16.778122] [0x00007b63558efc80] [info]    allow_multicolor_oneplate 1, allow_rotations 1 skip_modified_gcodes 0 avoid_extrusion_cali_region 0 loaded_filament_ids size 0, clone_objects size 0, skip_useless_pick 0, allow_newer_file 0
[2025-01-23 12:00:16.778133] [0x00007b63558efc80] [info]    plate_to_slice=0, normative_check=1, use_first_fila_as_default=0
[2025-01-23 12:00:16.778154] [0x00007b63558efc80] [info]    read model file:/home/debauer/projects/privat/generate-stl-files/stl/bin_z8_x3_y4.stl

[2025-01-23 12:00:16.806178] [0x00007b63558efc80] [debug]   TriangleMesh::repair() started
[2025-01-23 12:00:16.808988] [0x00007b63558efc80] [debug]   TriangleMesh::repair() finished
[2025-01-23 12:00:16.813599] [0x00007b63558efc80] [info]    object bin_z8_x3_y4.stl, id :5, from stl or other 3mf

[2025-01-23 12:00:16.813624] [0x00007b63558efc80] [info]    run:before load settings, file count=2

[2025-01-23 12:00:16.813643] [0x00007b63558efc80] [info]    operator():load setting file /tmp/process.json, with rule 1

operator(): unknown config type  of file /tmp/process.json in load-settings[2025-01-23 12:00:16.813917] [0x00007b63558efc80] [info]    record_exit_reson:441, saved config to result.json

run found error, return -5, exit...
[2025-01-23 12:00:16.813933] [0x00007b63558efc80] [info]    cli_callback_mgr_t::stop enter.
[2025-01-23 12:00:16.813937] [0x00007b63558efc80] [info]    cli_callback_mgr_t::stop not started before, return directly.
```

### Expected results

Generating gcode

### Project file & Debug log uploads

There is no Projekt.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

debauer, 2025-01-23:
After deep dive into the code this is my finding:
In the process and machine json is missing the type. So adding `"type": "process"` or `"type": "machine"` fix the loading. 
But now my process is not compatible with the printer and i need to figure this out. 

I think this is a bad way to handle this configurations. OrcaSlicer should by default add the type or split the --load-settings argument into 2 arguments. And! print a better error message. But this also applies to other strange error messages.

Any suggestions?

CodeMasterCody3D, 2025-02-07:
What I have found is that, orca-slicer uses the bambu-slicer cli the help menu has not been updated, I am in the process of trying to make a script to convert the .json file to ini 
But before I do that I need to go and get a working ini files from. Bambu slicer and then try and use it or change it up to match my json filament settings. 
Once I do I'll report back. 
I am trying to get a front end cli interface working with orcaslicer so I can have my local web server working with cli commands.

vl-201, 2025-03-13:
> What I have found is that, orca-slicer uses the bambu-slicer cli the help menu has not been updated, I am in the process of trying to make a script to convert the .json file to ini But before I do [...]

Did you get anything?

closed 2025-06-19 by github-actions[bot]