OrcaSlicer/OrcaSlicer #8161: Should "Wipe Distance" allow percentage? (Instead of only distance?)

Ergonomicmike, 2025-01-23:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

I don't know if this is similar to https://github.com/SoftFever/OrcaSlicer/issues/7326.

Or if this is an intended effect or a bug.

Orca 2.3.0-dev, Nightly as of January 23, 2025. 

Today I printed an [OpenVessel 2.0](https://www.printables.com/model/1152427-openvessel-20-the-yachty-a-practical-3d-printing-b/comments) alternative to the Benchy. I had "Wipe while retracting" enabled in Printer settings. (In the Extruder 1 tab.)

[image]

I used Adaptive Height. So the flow rate of filament is largest at the bottom of the model.

[image]

The model printed with a gash where the wipe takes place, where the flow rate is the largest. 

There was no gash anywhere else on the model.

You can see in the flow rate screen capture that the flow rate is "zero" in places. (Actually, it drops to 0.01.) So the print is printing as Orca commands.

[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

If this is not an intended effect, I'm wondering if, instead of only setting wipe distance in mm, it would be good to allow wipe "distance" to be in percentage (of the nozzle volume?), as is currently implemented for seam gap?

### Describe alternatives you've considered

As workarounds:

I can disable "Wipe on retracting."

I can lower the "wipe distance."

I can experiment with the "Retract amount before wipe."

### Additional context

I have used the PA tower to calibrate my Pressure Advance, and I believe that the PA is correct for this filament.

Ergonomicmike, 2025-01-24:
P.S. I had started with suggested values of Retract before wipe from @igiannakas [here.](https://github.com/SoftFever/OrcaSlicer/issues/5896#issuecomment-2323063586)

But I read in the Bambu wiki that I should set the Retract amount before wipe to zero.

70% or 0. That's quite a difference. (Unless I have it backward and this results in a change of 30%.)

Ergonomicmike, 2025-01-24:
And in case this might be related, 

https://forum.prusa3d.com/forum/prusaslicer/extruder-retract-amount-before-wipe-option-and-underextrusion/#post-705551

igiannakas, 2025-01-24:
Personally I use 70% wipe before retract on my voron and 1mm wipe move. Haven’t really had any issue with it since reworking that feature to fix a few extruder stalling bugs that are still present in bambu studio:

https://github.com/SoftFever/OrcaSlicer/pull/3094

https://github.com/SoftFever/OrcaSlicer/pull/3888

In general this feature is meant to control stringing. If you don’t have any, you can gradually reduce these two values to as close to 0 as possible for better seam performance (the wipe move if done with a nozzle that is not leaking can cause slight seam deformation as you’re wiping a dry nozzle over it).

Prusa slicer implements this a bit differently. With prusa you don’t control the wipe length, rather it’s calculated based on the current speed and retraction speed and retraction amount left. 

With orca, you can set these, but the retraction before wipe gets automatically larger if the distance taken to wipe within the constraints of the max retraction speed is smaller than what it would need to do the remaining retraction.

Take a look at the PRs above to better understand how it works :)

igiannakas, 2025-01-24:
If you could also attach me your project file I could take a look to see what’s happening specifically with the combination of settings

igiannakas, 2025-01-24:
What you will see if you set them too high are artefacts like this at the seam:

[image]

Notice how the seam next to the words benchy on the left is slightly smeared and slightly extended.

In general I would advise a high retract before wipe and a small wipe move to clean the nozzle. So something like 80% retract before wipe and 1mm or so of wipe to clean it a bit and eliminate stringing.

If you disable them you’re basically setting 100% retract before wipe and 0 wipe move if that makes sense?

Ergonomicmike, 2025-01-24:
Makes sense. I've been having the smeared seam on some of my prints. It seems to have started when I enabled Wipe on retract. Maybe I should go back to disabling it? (I noticed a gash big time on the PA Tower calib test, which went away as PA increased up the  Tower. But now I wonder if the calibration test should disable Wipe on retract for the test?)

But, as per my OP, 1 mm distance seems too much for this particular model and/or for where I've placed the seam. (Or is PA kicking in? So many variables, so little time.)

BTW, nice Benchy, I've never seen the lettering that distance on any that I've done. Am guessing that you used a 0.2 mm nozzle for that particular print?

igiannakas, 2025-01-24:
The PA test shouldn’t be affected really by this - your should be judging the corner performance as the seam will be pretty bad due to the PA being too low to start with and that defect propagating up the layers.

If you have wipe distance enabled, even with 0% wipe before retract it will retract before the wipe if you are printing fast. That is because if it wiped the full retraction amount during a very small wipe distance you would cause the extruder to stall and create a blob. 

Hence the primary tuning variable is the wipe distance and then the retract before wipe. Get the distance right so your seam is looking good and then either leave the retract at 0 and let it calculate it automatically or set a minimum % retraction value, if that makes sense?

Ergonomicmike, 2025-01-24:
Oops. Nevermind @igiannakas . I was playing around with the project file that I sent you. I disabled retract on wipe, but the problem remained. Then I realized that I had scarf joint seam enabled. 

I'll close this shortly.

igiannakas, 2025-01-24:
Ah no worries at all!!

closed 2025-01-24 by Ergonomicmike