OrcaSlicer/OrcaSlicer #8164: Build on macOS fails if libspnav is installed

phillc54, 2025-01-24:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0-dev

### Operating System (OS)

macOS

### OS Version

15.1.1 (Sequoia) 

### Additional system information

CPU: Apple M1

### Printer

None

### How to reproduce

1. Install libspnav with Homebrew
2. Compile latest from Github (b754734) using the script `build_release_macos.sh`

### Actual results

Build fails with the following message:
`** BUILD FAILED **
The following build commands failed:

CompileC /Users/phill/git/orcaslicer/build_arm64/build/libslic3r_gui.build/Release/Objects-normal/arm64/Mouse3DController.o /Users/phill/git/orcaslicer/src/slic3r/GUI/Mouse3DController.cpp normal arm64 c++ com.apple.compilers.1lvm.clang.1_0.compiler (in target 'libslic3r_gui' from project 'OrcaSlicer')
(1 failure)
`

### Expected results

Build should be successful.

### Project file & Debug log uploads

No log file, build unsuccessful

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

Alternate build using [these instructions](https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Mac%20OS.md) fails with the following message:
 `/Users/phill/orcaslicer/src/slic3r/GUI/Mouse3DController.cpp:21:10: fatal error: 'spnav.h' file not found
   21 | #include <spnav.h>
      |          ^~~~~~~~~
21 warnings and 1 error generated.
make[2]: *** [src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/Mouse3DController.cpp.o] Error 1
make[1]: *** [src/slic3r/CMakeFiles/libslic3r_gui.dir/all] Error 2
make: *** [all] Error 2
`

These are the contents of the Homebrew installed libspnav:
`$ ls -al /opt/homebrew/lib/libspnav*.*
lrwxrwxrwx  1 phill  admin  37 24 Jan 16:45 /opt/homebrew/lib/libspnav.a -> ../Cellar/libspnav/1.1/lib/libspnav.a
lrwxrwxrwx  1 phill  admin  41 24 Jan 16:45 /opt/homebrew/lib/libspnav.dylib -> ../Cellar/libspnav/1.1/lib/libspnav.dylib

$ tree /opt/homebrew/Cellar/libspnav/1.1
/opt/homebrew/Cellar/libspnav/1.1
├── INSTALL_RECEIPT.json
├── LICENSE
├── README.md
├── include
│   ├── spnav.h
│   ├── spnav_config.h
│   └── spnav_magellan.h
├── lib
│   ├── libspnav.a
│   └── libspnav.dylib
├── sbom.spdx.json
└── share
    └── pkgconfig
        └── spnav.pc
`

ShaneDelmore, 2025-01-24:
Why install libspnav?  I don't use it when I build on my M1 mac, and I don't see it referenced anywhere in orcaslicer build instructions. Have you seen this video with a walkthrough on building on a fresh system?  https://youtu.be/YiZvjZCoKbo?si=okEuZWfX8UIRQN_v

phillc54, 2025-01-24:
Because AFAIK it is required for building FreeCAD

phillc54, 2025-01-27:
After playing around a bit I managed to get it to build using the attached patch, it also appears to run correctly so far.

I am not really sure if this is the correct way around this issue though.

[macos_libspnav_build_fix.patch](https://github.com/user-attachments/files/18553140/macos_libspnav_build_fix.patch)

closed 2025-05-05 by github-actions[bot]