OrcaSlicer/OrcaSlicer #8165: Excluded Area for every Extruder on Multitool Printers

haunter1982, 2025-01-24:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

You always have to change printer profile for multitool prints if you want to use left or right or both extruders.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

For Multitool Printers with Dual Printheads and or IDEX it would be a great plus to have Excluded Bed Areas for every Extruder.

Workaround now is to have more printer profiles.

But I don't want to change the printer when I see it is needed to use support material or perhaps want to print in different colors.

### Describe alternatives you've considered

Give the Excluded Bed Area Setting for every extruder/tool perhaps like on SliCR-3D

[image]

[image]

[image]

Weeris102, 2025-02-09:
that's not orcaslicer?

haunter1982, 2025-02-09:
No it's Not. That's a example of how should be.
As SliCR-3D is also a fork Prusa/Superslicer.
That's why it is marked as enhancement