OrcaSlicer/OrcaSlicer #8166: Feature request: Visually represent scarf seam

tlhintoq, 2025-01-24:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

It would be really helpful if scarf seam and regular seam had different colors in gCode preview, especially when using `Conditional scarf joint` and its values leading to mix of seam types on the same print.

[image]

[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I'm trying to work through values for a good conditional scarf joint... but when I can't tell looking at the gCode where a normal seam and where a scarf seam should be, looking at good and bad points on the physical print tells me nothing.

Maybe keep regular seams white and scarf seams black?

henrivdr, 2025-01-25:
As a intermediate workaround, you can look at the Flow view.
The ramp up in flow along the scarf shows nicely.

tlhintoq, 2025-01-25:
> As a intermediate workaround, you can look at the Flow view. The ramp up in flow along the scarf shows nicely.

Thanks!  I'll give that a try

Wegerich, 2026-09-04:
Flow view is perfect for visualising the scarf. I'd suggest closing this ticket.