OrcaSlicer/OrcaSlicer #8171: optimization of idle movements

snakeRuSE, 2025-01-24:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

there is no problem, but if you hit a part, especially at high accelerations and jerks, you can break the extruder and print incorrectly.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

[image]

often when printing a large number of parts, the print sequence is from 1 to 15 (this is just an example) it looks like this. 1,2....15,1,2.
After part 15, the printer returns to part one, flying diagonally across the entire table. 

The simplify slicer had an "idle movement optimization" feature. With this function, after finishing printing part 15, the printer would start printing the next layer on part 15, and then go in reverse order.

henrivdr, 2025-01-25:
I'm curious to know how "Avoid crossing perimiters" would affect the path back to 1.
Would it move the long way around everything?
That could be a temporary workaround. (Excuse the pun 😬)

Stevetm2, 2025-01-25:
Avoid crossing walls does not work most of the time, I have it enabled and it crosses the plate like this, on travel moves, regularly.  Fixing the bugs in Avoid Crossing Walls would elevate this.  I believe there have been many tickets requesting this fix but they always go stale.

snakeRuSE, 2025-01-25:
A temporary solution may be to print the item as listed. In this case, the details in the list should be arranged in such a way that the first and last are located side by side.

[image]

igiannakas, 2025-01-28:
The reason this happens is to avoid printing hot filament over the just printed object that has not cooled down fully. This can lead to layers having uneven temperature states and as a consequence uneven contraction and expansion, creating visual artefacts on their external perimeters. Hence the print order is or should be maintained.

Basically this would make the benchy hull line problem worse.

snakeRuSE, 2025-01-28:
> The reason this happens is to avoid printing hot filament over the just printed object that has not cooled down fully. This can lead to layers having uneven temperature states and as a consequence [...]

It's a little weird. When one part is printed, is everything okay with its layers?

skyshard, 2025-03-19:
this seems related to https://github.com/prusa3d/PrusaSlicer/issues/11658 , which was fixed in [PrusaSlicer 2.9.1-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.1-alpha1) and could be cherry picked from https://github.com/prusa3d/PrusaSlicer/commit/bc36722d4b1d0407aa229b8d5da28f819816ea2d