OrcaSlicer/OrcaSlicer #8176: Crash while Export in Gcode

govorun17, 2025-01-24:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Win 11 24H2

### Additional system information

12th Gen Intel(R) Core(TM) i7-12700H 2.70 GHz
RAM 24,0 ГБ

### Printer

Ender 3 pro

### How to reproduce

1) Open stl file
2) Run export to gcode
3) Get crush

### Actual results

i've got log files with same string from time to time

[error]	2025-01-24 22:24:00.611832[Thread 0x00002b44]:Slic3r::GUI::PartPlate::calc_exclude_triangles:Unable to create exclude triangles

### Expected results

gcode file

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/18540923/log.zip)

[Flexi_Holding-Cat_Long_v1.zip](https://github.com/user-attachments/files/18540943/Flexi_Holding-Cat_Long_v1.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

### Anything else?

i think i got issue
save as 3mf and then slice

Noisyfox, 2025-02-09:
Could you please upload the 3mf file instead?

govorun17, 2025-02-09:
[Flexi_Holding-Cat_Long_v1.zip](https://github.com/user-attachments/files/18724636/Flexi_Holding-Cat_Long_v1.zip)
U r welcome

Noisyfox, 2025-02-09:
Odd.. Cannot reproduce the crash. Did you get the crash every time? Do you have special characters in the dest folder?

govorun17, 2025-02-09:
do you open .stl file?
i open from download folder
i've got crash just in case with .stl

govorun17, 2025-02-09:
after yours comment i can not reproduce crash :|

govorun17, 2025-02-09:
i think we can close issue, if i will cath it one more time, i'll reopen it with screenshots