OrcaSlicer/OrcaSlicer #8193: Creality K2 Full Support

danybp, 2025-01-25:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Looking to see if there is a timeline for full support on Creality K2. 

1. CFS support (similar to AMS)
2. Be able to upload to K2 SD card. (send vs print)
3. Camera feed under devices 
4. noticed other users mentioning issues with G-code for multi color printing

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

Full seamless support for K2 similar to Bambu printers.

bistory, 2025-01-26:
> ### Is there an existing issue for this feature request?
>
>     * [x]  I have searched the existing issues
>
>
> ### Is your feature request related to a problem?
>
> Looking to see if there is a timeline [...]

There have been fixes for K2 Plus in the latest builds ;-)

arushe0805, 2025-01-26:
> There have been fixes for K2 Plus in the latest builds ;-)

So the nightly builds have this already nice.
I looked the other day through recent commits and didn't see anything on it.

NateHackerman, 2025-02-18:
Device tab does not show camera. Port is 8000. To view on a web browser it is https://ipaddress:8000

YourSandwich, 2026-04-26:
Some firmware-side context on the K2 / K2 Plus CFS that may help slicer integration. I'm finishing a clean-room open-source reimplementation of the K2 Plus's compiled klipper extras (the Cython `.cpython-39.so` modules in `/usr/share/klipper/klippy/extras/` - the box wrapper, the RS485 wrapper, the filament rack, and friends). Plan is to release publicly soon as a reference design plus the reverse-engineering notes (RS485 protocol, decoded traces, architecture, full error / state tables).

A few things from the firmware side that may be directly useful for slicer-side CFS work:

### Multi-CFS support

CFS protocol runs at 460800 baud on the box RS485 bus. The address table allocates 4 slots, so the firmware models **up to 4 chained CFS boxes** at addresses `0x01..0x04`. Only verified single-box on hardware here, multi-box wiring is untested - but the protocol path is already there.

### `materialBoxs[].type`

In @MrTomek's payload: `type: 0` is a CFS box (the enclosed unit, temp + humidity sensor, 4 slots), `type: 1` is the external open spool rack (different protocol path, no temp/humidity readings). The JSON he posted matches that - the `type: 0` box has `temp` / `humidity` and 4 materials, the `type: 1` box has neither and one material.

### Slot indexing and tool numbers

Correction to an earlier version of this comment - tool numbers are a **3-char string** of the form `T<box><slot>`, where `<box>` is a 1-based digit `1..4` and `<slot>` is a letter `A..D`. Examples: `T1A` is box 1 / slot A, `T1C` is box 1 / slot C, `T4D` is box 4 / slot D. There is no `T0`. Internally the firmware does `(int(tnn[1])-1, ord(tnn[2])-ord('A'))` to get a 0-based `(box, slot)` pair.

Per-box slots map to a **bitmask** `A=0x01, B=0x02, C=0x04, D=0x08`. The bitmask is what gets sent on the RS485 frame's data byte for per-slot operations. The websocket payload's `materialId` is the 1-based index of the slot within the box.

### Multi-color tool change progression

The stage machine for the per-tool cut / retract / load / flush sequence is **driven from gcode macros** in the printer's `box.cfg`, not from Python and not from the websocket. The slicer-printer contract on multi-color is largely a macro/gcode contract. That is also why some of the multi-color issues people have hit need to be debugged against `box.cfg` on the printer, not against the slicer alone.

### Material database

CFS material types are persisted in JSON at `/mnt/UDISK/creality/userdata/box/material_database.json`. ~80 entries keyed by a 6-char hex code (e.g. `"000003"` = Generic PETG, `"006002"` = Hyper PETG). Each entry carries vendor, type name, color, min/max nozzle temp, target velocity, density. The `rfid` field on the per-slot websocket payload is exactly this key - so a slicer that reads or mirrors this file can resolve readable material names and temp ranges from the code.

### Box mode enum

`GET_BOX_MODE` returns a single ASCII digit:

```
'0' - IDLE         (box idle, ready for commands)
'1' - PRELOADING   (running the pre-load sequence on a fresh slot)
'2' - PRINTING     (active print, slot bound)
'3' - WRAPPERING   (wrap / pack-up phase)
'4' - ERR          (error state, needs reset)
'5' - TEST         (manufacturing / diagnostic mode)
```

A slicer can use this to gate when it is safe to send a tool change or a retract.

### Frame status byte

Every RS485 response frame from the box carries a status byte at offset 3. `0` is success, anything else is a structured failure. Decoded values:

<details>
<summary>RS485 status codes (click to expand)</summary>

```
0  - OK
1  - PARAMS_ERR        (bad parameters)
2  - CRC_ERR           (frame CRC mismatch)
3  - STATE_ERR         (command not valid in current state)
4  - LENGTH_ERR        (frame length wrong)
5  - EXTRUDE_ERR1
8  - EXTRUDE_ERR4
9  - EXTRUDE_ERR5
10 - EXTRUDE_ERR6
11 - EXTRUDE_ERR7
12 - EXTRUDE_ERR8
13 - EXTRUDE_ERR10
14 - EXTRUDE_ERR9
19 - RETRUDE_ERR1
20 - RETRUDE_ERR2
21 - RETRUDE_ERR3
22 - RETRUDE_ERR4
23 - RETRUDE_ERR5
25 - RETRUDE_ERR6
26 - RETRUDE_ERR7
34 - MOTOR_LOAD_ERR    (motor failed to load filament)
[... 25 lines cut ...]
key841 - cut error, cut sensor not detected, cutting not rebound
key843 - rfid error
key844 - pneumatic joint abnormal, may collapse
key845 - nozzle blocked
key846 - empty printing, box speed smaller than extruder
key847 - empty printing, material enwind
key848 - material error, may break at connections
key849 - retrude error, failed to exit connections
key850 - retrude error, unspecified, multiple connections triggered
key851 - retrude error, retrude but not trigger buffer empty limit
key852 - check extruder filament sensor and box sensor state
key853 - humidity sensor error
key855 - cut position error
key856 - no cutter
key857 - motor load error
key858 - eeprom error
key859 - measuring wheel error
key860 - buffer error
key861 - left rfid card error
key862 - right rfid card error
key863 - retrude error, retrude success but filament sensor still detected
key864 - extrude error, extrude but not trigger buffer full limit
key865 - retrude error, failed to exit connections
```
</details>

mjeanrichard, 2026-05-01:
@YourSandwich  That is great new! Where / how are you planning to release this?

Jacob10383, 2026-05-05:
> [@mjeanrichard](https://github.com/mjeanrichard) thanks, i was not aware. Different scope - jacobt replaces the whole firmware stack with his own. Mine is a 1:1 FOSS reference for the closed klipper_extras modules, decoded from the binaries and RS485 traces. The point is to have a reference base anyone can drop into stock, into a custom OS like his, or use as the starting point for their own improvements or clean-room rewrite.
> 
> My final goal is to install Arch Linux ARM on the K2, improve the firmware on top, rewrite parts in Rust, and make sure every feature of the original hardware and software is preserved. The touchscreen UI gets rewritten too - i already have a working concept for that.

Hey there, that is mine. Likewise I was not aware of your work until recently but sounds promising! Mine runs on reverse engineered replacements of the box, motor control, and  serial485 .SO’s
Since you said you are doing all the extras, I was wondering if you have made progress on PRtouch?

YourSandwich, 2026-05-05:
If there is interest I could release the files prematurely, currently refactoring function and methods to match the decompiled binaries and documenting, these are the semi-tested working ones for now:
```
  - box_wrapper.py
  - serial_485_wrapper.py
  - filament_rack_wrapper.py
  - load_ai.py + load_ai_extensions.py
  - filament_rack.py
``` 
Otherwise, I expect to be done in about 2-3 weeks with the full implementation.