OrcaSlicer/OrcaSlicer #8194: Add option to disable automatically dropping objects to bed

ansonl, 2025-01-26:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

 OrcaSlicer currently auto drops objects to the bed unless they are imported as a single nested object. 

I have multiple pieces in a file that need to be split and printed on different beds. OrcaSlicer  only lets them be imported as all separate objects or a single object. 

- Importing as separate objects drops them all to bed
- Splitting the single object in BS drops them to bed before I can Merge them into smaller collections.

In both cases the positioning of objects is lost.

OrcaSlicer  should add an option to disable automatically dropping objects to bed on import and splitting so that users can retain the object positioning.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Option in OrcaSlicer preferences to turn off automatically dropping objects to bed. 

### Describe alternatives you've considered

1. Separate the objects into files grouped by objects in Blender before importing but then you need 10 files or more for large 3MF files with lots of groupings. 

### Additional context

Use case is the highlighted pieces in the screenshot. each hexagon consists of 2 pieces for 4 pieces total which are prepositioned in the model at the right height. Right now there is no way to import all of them into OrcaSlicer in a single 3MF and then move the 2 hexagons (of 2 pc each) individually without dropping to bed. 
[image]

Sample file

[CA-sqrt-dual-transparent-v2-hex200scale-test.zip](https://github.com/user-attachments/files/18548798/CA-sqrt-dual-transparent-v2-hex200scale-test.zip)

primesun, 2025-02-10:
Linking to what I believe are basically the same issue:

- https://github.com/SoftFever/OrcaSlicer/discussions/6481
- https://github.com/SoftFever/OrcaSlicer/issues/2878
- https://github.com/SoftFever/OrcaSlicer/issues/724

And from the BambuStudio repo:
- https://github.com/bambulab/BambuStudio/issues/3491
- https://github.com/bambulab/BambuStudio/issues/3495

primesun, 2025-02-10:
I think this is the place in code that does that? Not 100% sure, only did a quick search to find this:

https://github.com/SoftFever/OrcaSlicer/blob/c21b044a9c566db272adaee7d88bfd7198c2a572/src/libslic3r/Model.cpp#L1625

hotellonely, 2025-02-11:
yes please make this happen. when exporting from fusion 360 it's always a pain.

J-W-Smith, 2025-02-21:
This should be an option to turn off, every other slicer I've used from Simplify3d on has had the "Drop to plate" as an option and not something that needs a hack to make work. Some parts print better with just supports. It has nothing to do with "Can't print parts in air", the supports will be added to anything that's not touching the build plate. It shouldn't be such a complicated issue.

MxBrnr, 2025-04-26:
I too would like to see this be an option, as would many others who have requested it on the Discord server. It seems to be asked about at least once per week.
Example:
https://discord.com/channels/1137181739773603922/1365823078738952313

ChemaTovi, 2025-05-21:
It must be an option like other slicers. I love orca and I want to print in air with supports. :C

WizardKnight, 2025-08-10:
Please add this option.

ansonl, 2025-09-25:
Until this improvement is ported over from BBS, you can manually group your objects in Blender under multiple Empty objects acting as groups. Then export from Blender as 3MF.