OrcaSlicer/OrcaSlicer #8196: Solid infill follows wall direction near holes and sharp corners

Jeevus3D, 2025-01-26:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Bambu X1C, Assorted Vorons

### How to reproduce

1. Open file with holes near walls or sharp corner features
2. Slice file
3. Observe solid infill behavior near features

### Actual results

Preview is shown, scroll to layer with solid infill near sharp features. Can see where the solid infill decides to follow direction of walls nearby instead of staying in a consistent 45 degree monotonic line. Issue is usually never an area larger than 8-10mm, though it follows through all solid infill top surface layers until the very top surface. Top surface is unaffected regardless of single wall setting on or off. 

### Expected results

Would expect the solid infill pathing to respect the solid infill direction set in the slicer settings, as well as the selected pattern. 

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/18549113/log.zip)

[Flexi-Rex Orca Issue - 1-26-25.zip](https://github.com/user-attachments/files/18549114/Flexi-Rex.Orca.Issue.-.1-26-25.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

### Anything else?

[image]

Noisyfox, 2025-01-26:
Disable "Detect narrow internal solid infill"

Jeevus3D, 2025-01-26:
THANK YOU!! Must have ticked it on messing with infill direciton.

closed 2025-01-26 by Jeevus3D