OrcaSlicer/OrcaSlicer #8199: Add Flatpak version to the Official Release

kiswarday, 2025-01-26:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

I recently upgraded my Linux OS (Linux Mint from 21.3 to 22.1) and the OrcaSlicer v2.2.0 Official Release appimage version doesn't work anymore for a missing library (libwebkit2gtk-4.0.so.37).
There are old open issues about this:
[https://github.com/SoftFever/OrcaSlicer/issues/8134](https://github.com/SoftFever/OrcaSlicer/issues/8134)
[https://github.com/SoftFever/OrcaSlicer/issues/4616](https://github.com/SoftFever/OrcaSlicer/issues/4616)

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Because the OrcaSlicer Appimage looks like doesn't ship with all libraries necessary, I had to install the flatpak-developing version to continue to use the program: [OrcaSlicer-Linux-flatpak_V2.3.0-dev.flatpak](https://github.com/SoftFever/OrcaSlicer/releases/download/nightly-builds/OrcaSlicer-Linux-flatpak_V2.3.0-dev.flatpak)

### Describe alternatives you've considered

1. Add alternative repositories to include the library missing => NO ("Frankenstein machine" problem)
2. Use a different slicer => NO
3. Use a different OS => NO

CS-Account, 2025-01-26:
Flatpaks truly are the most convenient and secure option for desktop apps on linux.

The sandboxing means you can be confident in what an app has access to, and flatpaks can easily be updated in 1 click or auto updated on most distros.

shininghero, 2025-01-26:
> Flatpaks truly are the most convenient and secure option for desktop apps on linux.
> 
> The sandboxing means you can be confident in what an app has access to, and flatpaks can easily be updated in 1 click or auto updated on most distros.

Convenient, yes.  But security has a caveat.  It's dependent on the app maintainer to bundle the latest versions of required libraries, and issue updates if any of them develops a CVE.  I, as the user, can't beat them to the punch and just update the system version of said libraries.

Personally, I'd like traditional .deb and .rpm files.  I'm used to those anyways.

kiswarday, 2025-01-26:
> Convenient, yes. But security has a caveat. It's dependent on the app maintainer to bundle the latest versions of required libraries, and issue updates if any of them develops a CVE. I, as the user, [...]

I don't know how the .deb (or .rpm) works internally, but I suppose, in my situation (and in general) the app could still not work because the library requested is missing from the Mint 22.1 repository.

I believed the appimage was the best solution for the Orca maintainers because appimages should be self contained and maintainers have full control on the app itself, but this is not in the reality.

I was lucky to find the flatpak dev version because otherwise, right now, I was unable to use Orca

ccalica, 2025-01-27:
> Convenient, yes. But security has a caveat. It's dependent on the app maintainer to bundle the latest versions of required libraries, and issue updates if any of them develops a CVE. I, as the user, can't beat them to the punch and just update the system version of said libraries.
> 
And this is a great example of the appimage requiring an external often updated dep provided by distro.  I had the same problem on Fedora.  Just installed the package and went on.  webkit should be updated outside the appimage or flatpak.

cochcoder, 2025-03-13:
Flatpak versions for both x86_64 and arm64 CPUs are included in official releases now!

timnolte, 2025-04-16:
Is OrcaSlicer now dropping AppImage support for FlatPak? I no longer see a nightly AppImage build. While some may prefer FlatPak I absolutely refuse to use them and don't appreciate being forced to do so.

neclimdul, 2025-04-18:
I don't have any insight into your question but probably not. More likely nightlies broke because expired key or something but also I wouldn't be surprised.

I used to agree and preferred an AppImage and in some cases kinda still do. But trying to come up with solutions to the webkit issue in various slicer AppImages has kinda opened my eyes to the problem for sufficiently large programs. Unless you're statically linking your entire OS into the app image, it has a hard dependency on the distro you run on being ABI compatible on all the libraries you leave out. 

That kinda defeats the entire purpose of building the container over a native package, especially since you can't install those requirements. You just hope they're there and correct and write about it in your README.

The main complaint against Flatpak tends to be all the dependencies you download. But that's the solution to actually working without being tied to the OS you're running on and so far in my experience it just works.

timnolte, 2025-04-18:
There are more problems with that when it comes to Flatpaks. The massive size that they take up and the amount of memory they consume because you are effectively shipping an entire OS with the app. Then there is the major pain of the sandboxing that happens and makes local data access a terrible experience. I have yet to have a good experience with Flatpaks apps. The Flatpak architecture needs a major revamp IMO.

CS-Account, 2025-04-18:
I don't think it's really approaching anything like shipping the OS with the app. You are just making sure it always has the dependencies it needs, so nothing ever breaks because somewhere along the chain something your app depended on changed in an unexpected way.

Regarding data access, typically Flatpak developers will have set access to be reasonable, and this is a great thing in terms of not giving every app full access to everything on your machine. They are given what is needed.

As for sandboxing, it's rather convenient in my experience and far safer than the way apps are typically allowed to do whatever they please as the user using them. Memory concerns, much like with storage concerns are just about less being shared between apps, but considering that in the modern world these are far less of a problem and I don't see it as a massive issue. You don't have a hundred instances of Flatpaks running, and using say, 250MB more ram isn't really a problem on a system with 16000 or more is it?

Of course everyone is entitled to their own opinions on efficiency, but at a certain point there is an idealistic element to it.

timnolte, 2025-04-18:
I honestly find it rather poor that what people feel is a minimum system requirement of 16GB of RAM being ridiculous these days. So I actually disagree when I have systems that are locked in at 8GB and I can't upgrade them. There really seems to be little regard for memory usage by Devs these days where they just assume everyone should just be running systems with 32GB and if not it's not their problem. 30 years ago Devs actually cared about such things.

I actually really don't care much for AppImages either as they are also bloated, but at least they can run in user space and don't need root permissions to install and run. The problem I see with Flatpaks and their sandbox is being unable to access removable drives. This is ridiculous and again is the same sort of garbage as expecting people to have 2TB+ system drives and never using removable storage.

The whole point of the Linux platform is to be open, and these poor choices and limitations go against that whole aspect IMO. If I wanted to work in a closed ecosystem I'd just use Windows or macOS and live with their restrictions.

eternal-sorrow, 2025-04-18:
> but at least they can run in user space and don't need root permissions to install and run

Flatpak also don't need root permissions to install. Just run it as `flatpak --user`.

timnolte, 2025-04-18:
> > but at least they can run in user space and don't need root permissions to install and run
> 
> Flatpak also don't need root permissions to install. Just run it as `flatpak --user`.

I'm perfectly aware of that. I wasn't talking about Flatpak not running that way. I was speaking to the fact that I prefer native apps over even AppImages, but I use them because they run in user space.

eternal-sorrow, 2025-04-18:
>The problem I see with Flatpaks and their sandbox is being unable to access removable drives.

Use FlatSeal or `flatpak override` to give access to the paths you need (and possibly take away the access to the paths you don't need).

My personal problem with flatpak is that most of the flatpaks give access to at least the whole user directory (and often the whole host filesystem). Which makes sandbox useless.

eternal-sorrow, 2025-04-18:
> I'm perfectly aware of that. I wasn't talking about Flatpak not running that way. I was speaking to the fact that I prefer native apps over even AppImages, but I use them because they run in user space.

I don't understand what are you trying to say here.

timnolte, 2025-04-18:
At this point there is no need to carry on. I'm hoping that there isn't a decision to drop AppImages and force Linux users to use Flatpak. If it comes to that then I'll have to get myself setup to build my own releases.