OrcaSlicer/OrcaSlicer #8200: Bambu printer certs will expire on non internet printers. Install certs from Orca.

cryptonym64, 2025-01-26:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Bambu lan only printers with internet blocked have expiring certs on them. Can Orca push new certs to the printer? Apparently they can use Bambu Slicer and support has mentioned an SD option, but when pressed replied that won't tell me how yet.

I would hope their forthcoming "solution" is not to simply install the crippleware firmware.

Here is what the last response was:

2025-01-26 15:58:31
Dear Customer,
Thank you for the reply.

We would like to kindly remind you that we will release and provide detailed certificate download and offline update guidance later. Thank you for your inquiry, and please stay tuned for updates.

### Which printers will be beneficial to this feature?

Others

### Describe the solution you'd like

Orca pushes replacement certs onto Bambu printers.

### Describe alternatives you've considered

Wait for Bambu to come up with another unworkable solution giving them more control.

### Additional context

Without non expired certs, only SD will remain usable to print from.

igiannakas, 2025-01-28:
This is a duplicate of https://github.com/SoftFever/OrcaSlicer/issues/8063

Please close the issue as at present information on this topic is limited to say the least. Cannot implement something that we dont know how it will work :)

cryptonym64, 2025-01-28:
My current guess is that the cert may be replaceable using the sslcert utility. I found a post using sslcert to pull the PEM off of a printer. I'm fine if this gets closed but the 8063 is not this specific issue.

It appears in my P1S that the cert is good until I think it was 2033, so if that is the only one, we have some time.