OrcaSlicer/OrcaSlicer #8204: Auto arrange not arranging all parts and mixes materials

crysxd, 2025-01-26:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

macOS

### OS Version

15.2

### Additional system information

MacBook Pro M1

### Printer

Voron 0

### How to reproduce

Open the attached file and change the printer to a v0. Press "Arrange all objects" with all options disabled (no rotation, no multiple materials on same plate, no align on Y).

Expected: All parts are sorted to plates, no same material on plate. This works with my other printer profiles (180x180, 235x235)

120x120:

### Actual results

Materials are mixed, not all parts on plates

[image]

### Expected results

Materials separated, all parts arranged onto plates

Works with 180x180:
[image]

### Project file & Debug log uploads

[Sample.3mf.zip](https://github.com/user-attachments/files/18551577/Sample.3mf.zip)
[log.zip](https://github.com/user-attachments/files/18551579/log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

thisnyjared, 2025-01-26:
As far as I know there is nothing that should make it separate materials, but I do think it should be added. I also think that you should try and see what your distance is set to for the parts to be separated by. But if it is small enough it dose sometimes choose a weird arrangement that dose not fit.

crysxd, 2025-01-27:
If you look at the second screenshot it separated all materials just fine, this is what this checkbox should do:
[image]

I also played around with the distance, but neither 0 (auto) nor 25mm (see below) solve the issue:
[image]

buzzhuzz, 2025-01-27:
@crysxd , could you please try latest dev build (for example here https://github.com/SoftFever/OrcaSlicer/actions/runs/12987442553 )? I just tried to reproduce issue using your 3mf file and it seems all working as intended: with spacing reduced to 1.3 Orca was managed to squeeze parts into 7 plates:

[image]

buzzhuzz, 2025-01-27:
Here is auto arrangement for Voron 0.1 selected from printer database:

[image]

Two parts on the bottom right not assigned to the plates since they are wider then a build plate size in that orientation and parts rotation is disabled.

Here how it looks like with parts rotation enabled:

[image]

crysxd, 2025-01-27:
I played around a bit more and found following things with  f11eb34 but did not find a significant difference to 2.2. I did notice though that using the system preset for the v0 did solve the issue so I systematically copied values from the v0 preset into mine until the issue was resolved. The bug seems to be triggered by the "prime tower enabled" option.

Given that I tell Orca to not mix materials...there shouldn't be a prime tower

Disabled:
[image]

Enabled:
[image]

closed 2025-05-05 by github-actions[bot]