OrcaSlicer/OrcaSlicer #8206: Better Flow Calibration

dMopp, 2025-01-26:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

No

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Hi guys, on maker world I found this one:
https://makerworld.com/en/models/893695#profileId-861958

And it’s damn good. It has the potential to replace the current flow rate calibration. The results are the same but it’s way easier to “read” .

Maybe you can contact that guy but I bet he would be happy to be the developer of the new standard :D

Ergonomicmike, 2025-01-27:
I thought that Orca's Flow Rate calibration was for top layer only?

For overall flow calibration, I suggest using a cali-flower type model. [Here's a version on printables.](https://www.printables.com/model/754877-xy-axis-calibrator-stepsmm-and-skew-correction)

I have used the latter to calibrate my overall flow and then I have used the former to set a nice top layer.

The bottom line is that I was able to print the makerworld test (actually, I printed his v5 beta version) and it showed a perfect flow rate.

igiannakas, 2025-01-28:
While this is useful to calibrate tolerances, there are more variables to tolerance other than just flow rate. For example how square the gantry is, material shrinkage and elephant foot to name a few.

Orca's flow rate calibration aims to produce the best visual appearance on top surfaces as this is the only place we can reliably "measure" the effect of flow rate. Adjusting flow rate to make parts fit is OKish, but it disregards other factors above, and may end up creating incorrect (too low) flow rate if the printer is not 100% mechanically sound.

dMopp, 2025-01-28:
But then I would suggest to rename flow calibration to “top flow calibration”. While this test here is more like general flow. I would even go a step further and put the top flow values into the printing profiles like the bottom surface. (Or put all kind of flows to the filament profiles, I don’t know what fits better).

So we have:
- Bottom Flow / First Layer - I found a test print for that: https://makerworld.com/en/models/896789
- General Flow (The one linked in first post)
- Top Flow (current flow test, even if there are some other options I like more, like: https://makerworld.com/en/models/189543?from=search#profileId-209504

Before that post here i was never aware its top flow only (but it makes sense… :D)

igiannakas, 2025-01-28:
Suggest if you could have a read here: https://ellis3dp.com/Print-Tuning-Guide/articles/extrusion_multiplier.html

It explains in detail the rationale for the test and why the aesthetics approach yields the optimal outcome for flow calibration.

It’s not top flow only, it adjusts the flow rate for the entire object in the calibration, but you’re using the top layer to judge what the best flow rate is. That is because the flow inaccuracies compound over multiple layers and show as artefacts on the top surface.

DVSVIDEO, 2025-01-28:
Currently in OrcaSlicer there are 3 options for setting the flow rate: 
- Global flow rate, can be set and saved individually for each material type 
- First layer flow rate (Global is 1 then 1.1 is 10% more i.e. 110%) 
- Top layer flow rate ( Global is 1 then 0.9 is 10% less i.e. 90%)

dMopp, 2025-01-28:
Well, more to read, more to understand :)

DVSVIDEO, 2025-02-05:
> I thought that Orca's Flow Rate calibration was for top layer only?
>
> For overall flow calibration, I suggest using a cali-flower type model. [Here's a version on [...]

This calibration is used to adjust the number of steps of the stepper motors and to correct geometric errors in the printer frame. What does it have to do with flow and its calibration?

Ergonomicmike, 2025-02-08:
@DVSVIDEO 

[According to the spreadsheet](https://docs.google.com/spreadsheets/d/1uTIV2XqXmuGRO1G5ICMuS5GCEh6hEW4SpkCMBcOr9DI/edit?gid=0#gid=0), some errors in the print are due to over or under extrusion.

See cells 14,15, 16 E, "wall thickness." The note in cell 19 refers to flow rate.