OrcaSlicer/OrcaSlicer #8214: Camera Style - Touchpad rotation sticking

cdunn95, 2025-01-27:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

Samsung Galaxy Book2 Pro 360

### Printer

Creality K1

### How to reproduce

1.) Press ctrl + p to open preferences.
2.) Enable `Touchpad` under General Settings -> `Camera style`
3.) Close Preferences
4.) Hold Alt in Preview or Prepare tab while hovering the model pane 
5.) Move cursor around
6.) Release Alt
7.) Continue to move cursor around, away from cursor position when Alt was released. (scrolling and panning are non-functional at this step)
8.) Tap (press and release) Alt, camera will snap to the position it would have been at had you instead continued to hold Alt in step 6.
9.) All functions (scrolling, panning, rotating) are now present again until you repeat step 4, 5, 6.

### Actual results

When attempting to rotate in touchpad camera settings, releasing Alt seems to result in a weird state where it is still rotating without updating the view until Alt is pressed again. All camera functions are disabled at this time. 

### Expected results

Rotation should only occur while Alt is held down. Releasing Alt should end the rotation mode and allow for other camera functions to be utilized. 

### Project file & Debug log uploads

[RotationBug.zip](https://github.com/user-attachments/files/18562673/RotationBug.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

buzzhuzz, 2025-01-27:
I'll look into that

buzzhuzz, 2025-01-27:
@cdunn95 , could you please provide screen capture showing how to reproduce this issue? I tried your steps on Linux machine, but was not able to reproduce issue.

cdunn95, 2025-01-28:
@buzzhuzz Sure, attached. 

Edit: Github won't allow me to upload the mp4 directly, so I had to place it inside of a zip.  Sorry about that.

[orca-bug-video.zip](https://github.com/user-attachments/files/18575882/orca-bug-video.zip)

cdunn95, 2025-01-28:
Trying again, converted to mov:

https://github.com/user-attachments/assets/9639a830-5bf5-4e08-a822-138e77f1c830

cdunn95, 2025-01-28:
I would like to point out that this may very well be attributed to [Windows Keyboard Events](https://learn.microsoft.com/en-us/windows/apps/design/input/keyboard-events), as I have experienced similar in another software I've helped with. My assumption would be that the switch in/out of the rotation may be linked only to the [KeyDown](https://learn.microsoft.com/en-us/uwp/api/windows.ui.xaml.uielement.keydown) event, and the [KeyUp](https://learn.microsoft.com/en-us/uwp/api/windows.ui.xaml.uielement.keyup) event. While they should be the events listened to in order to toggle on/off the rotation mode as a whole, the KeyUp event is actually only disabling the rendering of the rotation, but the rotation feature itself is still active. 

**This is just a prediction as someone that has not looked into the underlying code at all. Take it with the tiniest grain of salt.

buzzhuzz, 2025-01-29:
This is definitely Windows-specific behavior.

I've added debug logs on my test branch to make github build debug instance and ran on my Windows machine at the office.

Here are what happens (lines started with # are my comments):
```
# Here are we holding ALT and moving cursor rotating camera
on_key(ALT, down)
Mouse Event 8895 - Moving AltDown (1036, 691) - other
Mouse Event 8896 - Moving AltDown (1040, 690) - other
Mouse Event 8897 - Moving AltDown (1042, 689) - other
on_key(ALT, down)
Mouse Event 8898 - Moving AltDown (1044, 688) - other
Mouse Event 8899 - Moving AltDown (1046, 688) - other
Mouse Event 8900 - Moving AltDown (1049, 686) - other
Mouse Event 8901 - Moving AltDown (1050, 685) - other
on_key(ALT, down)
Mouse Event 8902 - Moving AltDown (1052, 684) - other
Mouse Event 8903 - Moving AltDown (1054, 684) - other
Mouse Event 8904 - Moving AltDown (1055, 683) - other
Mouse Event 8905 - Moving AltDown (1057, 682) - other
# Stop moving cursor and releasing ALT button
on_key(ALT, up)
### point1
# Due to some reason "window/widget leaving" event is generated by wxWidgets.
# This is happens on windows and not happens on linux machine
Mouse Event 8906 - Leaving (1059, 681) - other
# After some time - from few seconds up to a minutes we press ALT button again
# "widget/window enter" event is generated by wxWidgets
Mouse Event 8907 - Entering AltDown (1433, 1019) - other
# right after that next cursor move event generated, with ALT key pressed down.
# Here camera gets snapped to the new position since their reset have not been reset
# after ALT key release earlier
Mouse Event 8908 - Moving AltDown (1433, 1019) - other
# We releasing ALT key
on_key(ALT, up)
### point2
# here we may move mouse cursor immediately or wait some time:
# first move with no ALT key pressed shall reset camera state which is
# done here by calling to mouse_up_cleanup()
Mouse Event 8909 - Moving (1441, 1019) - other
on_mouse:4489
mouse_up_cleanup:5203
Mouse Event 8910 - Moving (1442, 1019) - other
Mouse Event 8911 - Moving (1469, 1019) - other
Mouse Event 8912 - Moving (1480, 1018) - other
Mouse Event 8913 - Leaving (1513, 1016) - other
```

On linux machines logs goes straight from point1 to point2 without leaving/entering window/widget events and no ALT down event spamming happens:
```
Mouse Event 2765 - Moving AltDown (221, 590) - other
Mouse Event 2766 - Moving AltDown (216, 592) - other
Mouse Event 2767 - Moving AltDown (211, 596) - other
Mouse Event 2768 - Moving AltDown (208, 597) - other
Mouse Event 2769 - Moving AltDown (202, 599) - other
Mouse Event 2770 - Moving AltDown (184, 604) - other
Mouse Event 2771 - Moving AltDown (172, 606) - other
Mouse Event 2772 - Moving AltDown (165, 607) - other
Mouse Event 2773 - Moving AltDown (161, 607) - other
Mouse Event 2774 - Moving AltDown (159, 608) - other
Mouse Event 2775 - Moving AltDown (157, 608) - other
Mouse Event 2776 - Moving AltDown (156, 608) - other
Mouse Event 2777 - Moving AltDown (155, 608) - other
on_key(ALT, up)
Mouse Event 2778 - Moving (155, 608) - other
on_mouse:4489
mouse_up_cleanup:5203
Mouse Event 2779 - Moving (156, 608) - other
Mouse Event 2780 - Moving (156, 608) - other
Mouse Event 2781 - Moving (157, 608) - other
Mouse Event 2782 - Moving (158, 608) - other
Mouse Event 2783 - Moving (159, 608) - other
Mouse Event 2784 - Moving (160, 608) - other
Mouse Event 2785 - Moving (160, 608) - other
```

I don't have windows machine at home, so debugging may take some time.

closed 2025-02-01 by Noisyfox