OrcaSlicer/OrcaSlicer #8218: Portion of object BELOW the print bed shouldn't be considered when deciding if object is `outside` or printable

tlhintoq, 2025-01-28:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0-dev

### Operating System (OS)

Windows

### OS Version

10proX64

### Printer

Elegoo giga but really all

### How to reproduce

Take a big odd shaped object... Then drop some of it through the bottom of the bed.  For example, you had a fail after 6 days and only need to print the remain 10 hours of the job and glue it on.

Try to position it for optimum location - but some of it is beyond the print bed - but only BELOW the print bed so it really shouldn't matter.  The part actually above the print bed is fine.

### Actual results

Object is considered `outside` or off any plate.  job can't be printed.

[image]

### Expected results

Only look at the job above the print bed.

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [x] Log file
- [x] Project file

discip, 2025-01-28:
Just use the **`Cut`** gizmo. 😊

tlhintoq, 2025-01-31:
> Just use the **`Cut`** gizmo. 😊

In some cases that's a good band-aid around the problem. But it would be nice to fix the problem.
If you've doing in-object painting of the object in colors - you just lost all that when you break up the object.

[image]

[image]

[image]

PR #8456 (2025-03-30): Only consider part above plate when checking for object outside
Fix #8218

Before:
[image]

After:
[image]
[image]

Note: there is still an issue that you still cannot slice if object below plate overlaps exclusion area. It won't be easy to fix and also will be computational costy, so I won't try to solve this in this PR yet.
[image]
commit 61efef9e (2025-03-30): Only consider part above plate when checking for object outside (#8456)

Fix #8218

Before:
[image]

After:
[image]
[image]

Note: there is still an issue that you still cannot slice if object below plate overlaps exclusion area. It won't be easy to fix and also will be computational costy, so I won't try to solve this in this PR yet.
[image]
closed 2025-03-30 by SoftFever