OrcaSlicer/OrcaSlicer #8220: Plug-in Support (Will revolutionize your slicer!!)

Pneumantic, 2025-01-28:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

First off, I just want to say thank you for everything you do. You guys are really something special in the 3D printing world and I appreciate your work. 

Something I feel should be one of the main priorities of the program is the ability to integrate Plugins. Almost all open-source projects that really gain a massive amount of traction are the ones that allow the community to provide support as easily as possible while not changing the main structure of the main project. Some examples of easy integration:

- Visual Studio Code Plug-ins Tab
- Sudo apt install commands
- Raspberry Pi OS installer
- Arduino Library Manager
- Arduino Board Manager
- Runlite Plug-in store
- Pip install
- VLC Media Player
- OBS Studio

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

These are just a few of the major users on the market that may be known. I would assume most of you at least know of VS codes plugin system. Something like this could quite literally revolutionize the slicer. You already have some people like Tentech on Youtube who is attempting to integrate through back door means. It seems like a lot of this is possible to be integrated. If this was done, a lot of the requests you see day to day will completely tank. There are dozens of features I would love to make but I am simply not willing to go out of my way to interface with how it is right now.    

Link to Tentech:
- https://www.youtube.com/@tengertechnologies

### Describe alternatives you've considered

The only way to perform most of these actions is by possibly editing these options. Again, a lot of this is filler and isnt really that good for modularity or if you want to manage the installed packages.

1. Post-processing scripts box in Others Tab
2. Begining/end/extruder gcode editing

### Additional context

The idea is that it would work like VS code almost. You would have a tab for a marketplace and you can search for various plugins which allow new features to be added.

Image of the VS marketplace as an example:
[image]

How it could be integrated:
[image]

vgdh, 2025-01-29:
I wish that was a simple thing.

paulo-granthon, 2025-01-30:
I'd love that and am open to contribute if there's any initiative on that.

I believe that a plugin system is something that would benefit the project a lot, making it easier for advanced users to implement the features they want but wouldn't really be viable for merging into the main branch, and for others to pick and choose increments that would benefit them without necessarily (again) including them for everyone else.

Not sure on the "Plug-ins" tab in Process. Maybe a dedicated master tab at the top, and for the parameters just allow the plugins to create new tabs with custom parameters.

Pneumantic, 2025-02-02:
> I wish that is a simple thing.

Could you give information on what the most difficult aspect would be? I dont know much about your guys back-end, but a filler bandaid could possibly be to have you import a repo URL  and then it fills the post processing slot and allows you to remove or add a feature by definitions the plugin developer creates. (So like adding a check box with a label they give, then they have to put in the file associated as they used in post processing). But I don't know how a lot of the slicer is coded. Simply adding a new tab may not be trivial or adding a new line for a parameter. What kind of info can you provide?

Pneumantic, 2025-02-02:
> Not sure on the "Plug-ins" tab in Process. Maybe a dedicated master tab at the top, and for the parameters just allow the plugins to create new tabs with custom parameters.

Issue I find with that is scripts that you want to be machine specific. Having a master tab would direct users to use it in control of all aspects. Integrating a million new tabs would create a lot of clutter and possibly integrating them into existing tabs creates a lot of extra coding challenges. What I was stating is hopefully just to get a ball rolling. If Orca gets anything remotely like this, they will most likely see an increase in contributor applications. Maybe down the road something like that would work but my post is trying to cut out as many extra variables as possible.

sammcj, 2026-06-12:
A plugin system would be amazing, and would make contributing to the project without having to modify the core significantly easier (I'm assuming this would hopefully reduce review / support overheads as well but maybe I'm being overly optimistic?).

An example of this recently was where someone figured out how to make overhangs printable, they forked the whole project (https://github.com/dennisklappe/OrcaSlicer-WaveOverhangs) - this would have been a lot nicer as a plugin of some sort.

TommasoLencioni, 2026-09-18:
Here it is: https://www.youtube.com/watch?v=CHbVJz3LYrY

thelegendtubaguy, 2026-11-08:
Ask and ye shall receive.  Please close the issue, plugins are in 2.5.0.