OrcaSlicer/OrcaSlicer #8221: Gaps in Walls

G3Dmaker, 2025-01-28:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.2.0 & 2.3.0-dev

### Operating System (OS)

Linux

### OS Version

Linux Mint 21.3 Cinnamon

### Additional system information

CPU: Raptor Lake-H (Core i9)
RAM: 32GB
Graphics: Intel driver: i915

### Printer

FireXX (DIY Printer) Marlin2

### How to reproduce

1. start: OrcaSlicer_Linux_V2.2.0.AppImage
2. load an stl-File
3. Slice
4. show the preview

### Actual results

There are spontaneous gaps in several walls.

[image]

[image]

[image]

### Expected results

Prusa Slicer

[image]

### Project file & Debug log uploads

[DusenReiniger.zip](https://github.com/user-attachments/files/18575322/DusenReiniger.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

igiannakas, 2025-01-28:
this is caused by the fan speedup time. set it to 0 and its will resolve.

However this shouldn't be happening, so its a bug indeed.

Noisyfox, 2025-02-09:
Seems like a rendering issue though, since the gcode is correct.

Noisyfox, 2025-02-09:
It's acutally caused by some iusse with the fan mover that it will break when using absolute E. I'll take a look. In the mean time, use relative E as a workaround.

PR #8359 (2025-02-15): Fix Fan mover when absolute E is used
Fix #8221

[image]
closed 2025-02-15 by Noisyfox