OrcaSlicer/OrcaSlicer #8227: Orca Slicer consistently crashes when slicing complex models with efficiency cores enabled (E cores)

drcircuit, 2025-01-28:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 11 PRO 23H2

### Additional system information

CPU: Intel i9 13800K
RAM: 128GB Corsair DDR4 
GPU: ASUS TUF Gaming RTX 4090
Orca Slicer will consistently trigger errors and crash when slicing complex models (>500000 triangles) with efficiency cores enabled.

We get errors and crashes, even BSODs when slicing complex organic models with my setup. 

However: Disabling e-cores in the BIOS or setting a permanent affinity for Orca Slicer to avoid the e-cores mittigates the problem.

Example model: https://www.thingiverse.com/thing:4401649#google_vignette - the helmet

This model will crash consistently, between clean boots and while running the system a while.

This is one example exception from the event viewer:
+ System 

  - Provider 

   [ Name]  Application Error 
   [ Guid]  {a0e9b465-b939-57d7-b27d-95d8e925ff57} 
 
   EventID 1000 
 
   Version 0 
 
   Level 2 
 
   Task 100 
 
   Opcode 0 
 
   Keywords 0x8000000000000000 
 
  - TimeCreated 

   [ SystemTime]  2025-01-28T00:21:05.4976141Z 
 
   EventRecordID 100809 
 
   Correlation 
 
  - Execution 

   [ ProcessID]  28668 
   [ ThreadID]  4612 
 
   Channel Application 
 
   Computer DESKTOP-8P2MMED 
 
  - Security 

   [ UserID]  S-1-5-21-1964434880-2946729229-652945515-1001 
 

- EventData 

  AppName orca-slicer.exe 
  AppVersion 1.0.0.0 
  AppTimeStamp 671e6394 
  ModuleName OrcaSlicer.dll 
  ModuleVersion 0.0.0.0 
  ModuleTimeStamp 671e6388 
  ExceptionCode c0000005 
  FaultingOffset 0000000000b22fd5 
  ProcessId 0x2bdc 
  ProcessCreationTime 0x1db711a7e1b6add 
  AppPath C:\Program Files\OrcaSlicer\orca-slicer.exe 
  ModulePath C:\Program Files\OrcaSlicer\OrcaSlicer.dll 
  IntegratorReportId 4c0514e7-c24c-46ca-9ae9-01eaefe93291 
  PackageFullName  
  PackageRelativeAppId  

However I have experienced a direct crash with just a clean exit from the process, and a KMODE BSOD exception as well.

It seems there is some issue with regards to syncing distributed loads across P vs E cores, maybe it is a timing issue, but I will leave it to you to decide. 

### Printer

CR-10 MAX / K1 MAX

### How to reproduce

1. Have a system as described
2. Download model
3. Slice
4. Observe crash

### Actual results

Disabling e-cores in BIOS or setting CPUID affinity in Windows mitigates the issue.

### Expected results

Allow Orca Slicer to utilize all cores and resources available without crashing.

### Project file & Debug log uploads

Just download the version of Orcaslicer and model on an analog system and try.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ZackySteel, 2025-02-03:
I have gotten consistent crashes as well, with only maybe 10% of models slicing without error and/or crash. I've tried the few troubleshooting methods I saw online, including reducing OrcaSlicer's priority, but it didn't work for me (I later saw a comment on the thread saying that this workaround doesn't work at all for 2.2.0). I have tried the 2.2.0 nightly release along with every older version (including betas) back to version 2.1.0, but continue to experience the same crashes/errors when slicing complex models.

For me, at least, this issue is the highest-priority in-terms of what needs to be fixed because OrcaSlicer is effectively unusable regardless of whether I am on my usual work laptop or my significantly more powerful home desktop.

Minecreator200, 2025-02-05:
Can you differentiate if it's because of the large triangle count (what numbers are we talking here?) or because of your hardware or device driver setup?

Noisyfox, 2025-02-09:
Try pin the app to P cores only and see if it still crashes.

Noisyfox, 2025-02-09:
Ok cool. So at least we have a workaround. Unfortunately I don't have an intel CPUs so I couldn't really debug this issue.

HerrDerLage7, 2025-03-01:
im an full AMD (r5 6500x + rx6900xt) user. had the same issue.
go to graphic seetings in windows, add orca slicer and choose your graphicscard on highest.
no issues anymore...
(german user tries to explain in english)

ElectricFox115, 2025-05-27:
Can confirm on an Intel 14900K, 128 GB. Orca is basically unusable with the E-cores on: just slicing will crash about 70% of the time, and if it survives, moving the layer slider will, without exception, kill it.

Switching affinity to cores 0 through 7 works, but there's a distinct hit in performance - it feels choppy. Attaching an error log.

[debug_Tue_May_27_13_56_25_2144.log](https://github.com/user-attachments/files/20453380/debug_Tue_May_27_13_56_25_2144.log)

closed 2025-09-03 by github-actions[bot]