OrcaSlicer/OrcaSlicer #8229: [Bounty: $250] Actual official website

johnny-bit, 2025-01-28:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Currently there's no official website and the ones that are showing up when someone unfamiliar with orca are not the official repo downloads.

I'd like to offer a bounty of $250 for anybody that creates an official OrcaSlicer website OK'd by @SoftFever.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

A repo named `softfever.github.io` needs to be created under @SoftFever's account and set up with an official domain name.

I 'd suggest making it in a way that would offer users quick download and maybe docs for orca.

@Pilotmatt over at orca-dev-chat on discord suggested using docusaurus.

Also - @Pilotmatt and several other people own domains registered in OrcaSlicer's name, so it's up to @SoftFever to choose any from that list. Here's the list shared by @pilotmatt on discord: 

[image]

(personally i'm leaning towards `orcaslicer.io`)

Also - over on same discord `@justinlevinedotme` purchased `.wiki` so that's quite a number avail.

### Additional context

Just to make sure: I'd like the website to be official, OK'd by @SoftFever on domain name of @SoftFever's choosing. Even rudimentary domain is ok.

PilotMatt, 2025-01-28:
To be clear I am happy to transfer any or all of those domains to someone; their registration expires August 2025.

KodeSage, 2025-01-29:
do you have a design crafted already or should I also come up with the design? @johnny-bit @PilotMatt

johnny-bit, 2025-01-29:
Up to you, as long as @SoftFever approves it and it becomes official.

WimVDK, 2025-01-29:
Well I am a hobby designer, have access to domain registration and servers. I work primarily with the Joomla CMS. No idea if that would work for a scenario such as this though

johnny-bit, 2025-01-29:
CMS like Joomla wouldn't work on github pages. 

Learn more on: 
- https://pages.github.com/  
- https://docs.github.com/en/pages/getting-started-with-github-pages/about-github-pages

emmakun, 2025-01-29:
There are some Github-based CMS for static websites, like [Decap](https://decapcms.org/) or [Tina](https://tina.io/). There are also Hugo, Jekyll, Gatsby and other Static Site generators that work well. Netlify could be an option for hosting a small official website.

KodeSage, 2025-01-29:
> Up to you, as long as [@SoftFever](https://github.com/SoftFever) approves it and it becomes official.

alright, I'm looking forward to it
geared towards building something amazing that you and @SoftFever would like.

johnny-bit, 2025-01-29:
> building something amazing that you and [@SoftFever](https://github.com/SoftFever) would like.

My only requirement for this bounty is that the site becomes official which obviously requires only @SoftFever's approval.

if it becomes a github repo, if I don't like something I'd be able to post an issue about it and if I would like to fix something I could post a PR 😄

SoftFever, 2025-01-30:
Hi all,

I will go through the thread tomorrow as I’m having trouble accessing
GitHub at the moment, it's late night here..

Luckily, I’ve registered the "orcaslicer.com" domain.
I definitely need help to create an official website and would prefer to
host it on GitHub too :)

KodeSage, 2025-01-30:
> Hi all,
> 
> I will go through the thread tomorrow as I’m having trouble accessing
> GitHub at the moment, it's late night here..
> 
> Luckily, I’ve registered the "orcaslicer.com" domain.
> I definitely need help to create an official website and would prefer to
> host it on GitHub too :)
> […](#)

alright sure, do let me know if you need my help in setting up the website.
I'm available to build it.

epoiate, 2025-02-01:
Hi!

I'd like to help to develop the site, I have experience with Angular, but I'm willing to learn React too.

PS: It would be really cool to use the .xyz domain for this

SoftFever, 2025-02-06:
Hi everyone,

It’s been a while, but I’ve finally managed to find a slot and set up the framework.

With the help of DeepSeek R1, I’ve created a simple website to start with.
The source code for the website can be found here: [GitHub Repository ](https://github.com/OrcaSlicer/orca-website).
The website is now live and can be accessed at: [orca-slicer.com ](https://www.orcaslicer.com/).

It’s hosted on GitHub Pages, using only HTML/CSS/JavaScript for simplicity.

I jumped on the AI hype train and had a fun time experimenting with DeepSeek R1—haha! 
The webpage was created entirely by AI. While it might not be perfect yet, I think it's a good start! 😊

If anyone has web design experience and would like to help improve it, please let me know! ; )

WimVDK, 2025-02-06:
you mean orcaslicer.com ;) that hyphen isn't in the url. Better edit. 

That said? Improve? I don't see anything to improve other than maybe (unless already done) a section to download the latest beta/RC/nightly alongside the stable. Unless already done it would be excellent if it automatically lists the new stable (eg 2.3.0) unless that's already done at the backend.

When finalized, ping me. I'll update the reddit community sub info.

SoftFever, 2025-02-06:
> you mean orcaslicer.com ;) that hyphen isn't in the url. Better edit.
>
> That said? Improve? I don't see anything to improve other than maybe (unless already done) a section to download the latest [...]

Nice catch! Fixed

SoftFever, 2025-02-07:
> Hey [@SoftFever](https://github.com/SoftFever) this looks like a great start!
> 
> Would you like to give [@KodeSage](https://github.com/KodeSage) (and maybe others) a chance to improve it before closing the issue?

Absolutely!
The current website is far from good.
@KodeSage or anyone with web design experience and an interest in improving the website, your contributions are most welcome.
If preferred, I can create a sub Discord channel to continue the discussion there.

KodeSage, 2025-02-07:
> > Hey [@SoftFever](https://github.com/SoftFever) this looks like a great start!
> > Would you like to give [@KodeSage](https://github.com/KodeSage) (and maybe others) a chance to improve it before closing the issue?
> 
> Absolutely! The current website is far far from good by any means. [@KodeSage](https://github.com/KodeSage) in case you have interest in improving the website, your contributions are most welcome. If preferred, I can create a sub Discord channel to continue the discussion there.

Hi @SoftFever , the website is good but I will love to improve on it too 😊.
here is my discord handle - @0xkodesage

KodeSage, 2025-02-14:
GM @johnny-bit @SoftFever based on this improvements , the updated version will be pushed up before wednesday next week.
Trust me, you would love the new look of the website 😊.

[image]

Here is a snippet of the new loook. Anctipate for the full and available website

SoftFever, 2025-02-14:
> GM @johnny-bit @SoftFever based on this improvements , the updated version will be pushed up before wednesday next week.
> Trust me, you would love the new look of the website 😊.
> 
> [image]
> 
> Here is a snippet of the new loook. Anctipate for the full and available website

This is bad ass!

johnny-bit, 2025-02-14:
Hey @SoftFever - if you don't mind, please create website-dedicated sub channel on discord :)

SoftFever, 2025-02-16:
> Hey [@SoftFever](https://github.com/SoftFever) - if you don't mind, please create website-dedicated sub channel on discord :)

Hi Johnny,
I checked the Discord server, I think the dev forum is a perfect place. Feel free to create a thread there.
[image]

WimVDK, 2025-02-19:
That looks crazy! I'll make an announcement/pinned post in the official subreddit once this has been merged

johnny-bit, 2025-03-13:
This is great! thanks @KodeSage and @SoftFever 
Now Orca has official website: orcaslicer.com
and the repo https://github.com/OrcaSlicer/orca-website/ is for anyone wishing to help :)
The bounty was awarded as per discussion with SoftFever.
There were discussions over at discord... Looks like for 2.3 release Orca will have official website and for 2.4 maybe even nice docs :)

closed 2025-03-13 by johnny-bit