OrcaSlicer/OrcaSlicer #8236: PORTABLE WIN VERSION FLAGS AS MALICIOUS ON VIRUSTOTAL

bubblesqueak24, 2025-01-29:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

PORTABLE WIN VERSION FLAGS AS MALICIOUS ON VIRUSTOTAL

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

PORTABLE WIN VERSION FLAGS AS MALICIOUS ON VIRUSTOTAL

DVSVIDEO, 2025-02-01:
Why not write to the makers of Virustotal?

Minecreator200, 2025-02-05:
It's very unlikely that those contain malware if the regular installer doesn't - right? But you can always build the releases yourself, if you really want to.