OrcaSlicer/OrcaSlicer #8237: Ironing support interface

YZcat2023, 2025-01-29:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Although using pla filaments as support interface material when printing petg parts is already breathtaking, i assume adding ironing effects will give a even better result.

[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Adding a option to enable ironing support interfaces

### Describe alternatives you've considered

A 3rd postprocessing script

YZcat2023, 2025-02-01:
err I enhanced my postprocessing program to do this. https://github.com/YZcat2023/MKPSupport. hope this function can  be integrated soon

ShaneDelmore, 2025-02-05:
Can you share some pictures of the improvement resulting from this change?