OrcaSlicer/OrcaSlicer #8241: Orca 2.2 - All custom printer and filament configs gone

drkline, 2025-01-29:
### Discussed in https://github.com/SoftFever/OrcaSlicer/discussions/8235

<div type='discussions-op-text'>

<sup>Originally posted by **drkline** January 29, 2025</sup>
I just opened up Orca and all of my custom filament and printer configs are gone.  So I also lost all of my custom start and stop gcode configs also.  Is there any way to recover these?  I'm really p****d at the thought that I may need to recreate all of this.  Thoughts? Help?  

Thanks</div>

buzzhuzz, 2025-01-29:
Could you please check if next launch will put their back. It might be the same issue as https://github.com/SoftFever/OrcaSlicer/issues/8094

You also could check latest dev build (for example that one https://github.com/SoftFever/OrcaSlicer/actions/runs/13017577998) to see if it helps.

drkline, 2025-02-01:
[image]

This is what I'm seeing.  There is a current config file and a backup.  It looks like there is some sort of config file from 1/24/2025 but it seems to be a corrupted file.  Here is the contents of that file.

{
    "app": {
        "allow_ip_resolve": true,
        "associate_3mf": true,
        "associate_step": false,
        "associate_stl": false,
        "auto_arrange": true,
        "auto_calculate": true,
        "auto_calculate_when_filament_change": true,
        "backup_interval": "10",
        "backup_switch": true,
        "camera_navigation_style": "0",
        "check_stable_update_only": false,
        "curr_bed_type": "3",
        "dark_color_mode": "1",
        "default_page": "1",
        "developer_mode": false,
        "download_path": "C:\\Use

drkline, 2025-02-07:
It looks like there are 2 user folders in there.  The 109056693 folder has all of my old correct settings but it appears that the application is only seeing what is in the Default folder.

[image]

drkline, 2025-02-07:
A few other things that I have noticed is that 1.) The Home screen is blank 2.) The application takes a minute or 2 to open and 3.) If I have it open on one of my other monitors it can drastically effect the performance of video applications on one of my other monitors.

drkline, 2025-02-07:
So, after doing some finagling with the folder names, I was able to get my profiles back and it now opens faster as well.  It seems that something went awry and the folders became corrupt.  All apears to be operational again.  I recommend that someone on your end investigate why this could have ocurred.

closed 2025-02-07 by drkline