OrcaSlicer/OrcaSlicer #8243: OrcaSlicer crashes on connecting to Klipper UI, and gets (almost) permanently corrupted - Linux, NixOS 24.11

TYTheBeast, 2025-01-29:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Linux

### OS Version

NixOS 24.11

### Additional system information

Kernel: Linux 6.6.72
Uptime: 1 day, 11 hours, 20 mins
Packages: 3520 (nix-system), 2136 (nix-user), 5 (flatpak)
Shell: fish 3.7.1
Display (DELL 1907FPV): 1024x1280 @ 75 Hz in 19" [External]
Display (27E1QA): 2560x1440 @ 144 Hz (as 2048x1152) in 27" [External] *
DE: KDE Plasma
WM: KWin (Wayland)
WM Theme: Layan
Theme: Breeze (Layan) [Qt], Layan-Dark [GTK2/3/4]
Icons: Papirus-Dark [Qt], Papirus-Dark [GTK2/3/4]
Font: Noto Sans (10pt) [Qt], Noto Sans (10pt) [GTK2/3/4]
Cursor: GoogleDot-Black (24px)
Terminal: .konsole-wrappe
CPU: AMD Ryzen 5 7600X (12) @ 5.45 GHz
GPU: AMD Radeon RX 7900 XT [Discrete]
Memory: 11.51 GiB / 31.09 GiB (37%)
Swap: 1.41 GiB / 33.20 GiB (4%)
Disk (/): 268.85 GiB / 598.00 GiB (45%) - btrfs
Local IP (wlp10s0): 192.168.1.22/24
Locale: en_US.UTF-8

### Printer

Neptune 4

### How to reproduce

1. Open OrcaSlicer
2. Go to Prepare or Preview
3. Click the little WiFi icon near the printer's edit button
[image]
4. Enter the various details required: IP, Device UI, API Key
5. Click Ok
6. OrcaSlicer is now corrupted until `~/,config/OrcaSlicer` and `~/.local/share/orca-slicer` are deleted or cleared.

### Actual results

A beautiful SEGFAULT appears if trying this through the terminal, oh the lovely landscape of C++!

### Expected results

I would expect OrcaSlicer to allow me to send files to my printer.

### Project file & Debug log uploads

[OrcaSlicer Bug Report.zip](https://github.com/user-attachments/files/18589616/OrcaSlicer.Bug.Report.zip)
[log.zip](https://github.com/user-attachments/files/18589610/log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

### Anything else?

[Here](https://youtu.be/sRWFMtdOOnk)'s a link to a demonstration / official video bug report for this issue.

eras, 2025-02-02:
I  am facing the same issue when adding an OctoPrint/Klipper connectivity to OrcaSlicer.

I did a bit debugging by compiling it with debug info enabled and I got this backtrace with `gdb`:

```
Thread 1 "orcaslicer_main" received signal SIGSEGV, Segmentation fault.
webkitWebViewBaseEnterAcceleratedCompositingMode () at ./Source/WebKit/UIProcess/API/gtk/WebKitWebViewBase.cpp:2938
Download failed: Invalid argument.  Continuing without source file ./build-soup2/./Source/WebKit/UIProcess/API/gtk/WebKitWebViewBase.cpp.  
2938	./Source/WebKit/UIProcess/API/gtk/WebKitWebViewBase.cpp: No such file or directory.
(gdb) where
#0  webkitWebViewBaseEnterAcceleratedCompositingMode(_WebKitWebViewBase*, WebKit::LayerTreeContext const&) ()
    at ./Source/WebKit/UIProcess/API/gtk/WebKitWebViewBase.cpp:2938
#1  0x00007ffff276fd38 in WebKit::WebPageProxy::enterAcceleratedCompositingMode(WebKit::LayerTreeContext const&) ()
    at ./Source/WebKit/UIProcess/WebPageProxy.cpp:10741
#2  0x00007ffff28a9f4e in WebKit::DrawingAreaProxyCoordinatedGraphics::enterAcceleratedCompositingMode(WebKit::LayerTreeContext const&) ()
    at ./Source/WebKit/UIProcess/CoordinatedGraphics/DrawingAreaProxyCoordinatedGraphics.cpp:237
#3  0x00007ffff24192a0 in _ZZN3IPC18callMemberFunctionIN6WebKit16DrawingAreaProxyES2_FvmRKNS1_16LayerTreeContextEESt5tupleIJmS3_EEEEvPT_MT0_T1_OT2_ENKUlDpOT_E_clIJmS3_EEEDaSI_ () at ./Source/WebKit/Platform/IPC/HandleMessage.h:135
#4  _ZSt13__invoke_implIvZN3IPC18callMemberFunctionIN6WebKit16DrawingAreaProxyES3_FvmRKNS2_16LayerTreeContextEESt5tupleIJmS4_EEEEvPT_MT0_T1_OT2_EUlDpOT_E_JmS4_EESA_St14__invoke_otherOSC_DpOT1_ () at /usr/include/c++/12/bits/invoke.h:61
#5  _ZSt8__invokeIZN3IPC18callMemberFunctionIN6WebKit16DrawingAreaProxyES3_FvmRKNS2_16LayerTreeContextEESt5tupleIJmS4_EEEEvPT_MT0_T1_OT2_EUlDpOT_E_JmS4_EENSt15__invoke_resultISA_JDpT0_EE4typeEOSA_DpOSM_ () at /usr/include/c++/12/bits/invoke.h:96
#6  _ZSt12__apply_implIZN3IPC18callMemberFunctionIN6WebKit16DrawingAreaProxyES3_FvmRKNS2_16LayerTreeContextEESt5tupleIJmS4_EEEEvPT_MT0_T1_OT2_EUlDpOT_E_S9_JLm0ELm1EEEDcOSA_OSC_St16integer_sequenceImJXspT1_EEE () at /usr/include/c++/12/tuple:1852
#7  _ZSt5applyIZN3IPC18callMemberFunctionIN6WebKit16DrawingAreaProxyES3_FvmRKNS2_16LayerTreeContextEESt5tupleIJmS4_EEEEvPT_MT0_T1_OT2_EUlDpOT_E_S9_EDcOSA_OSC_ () at /usr/include/c++/12/tuple:1863
#8  IPC::callMemberFunction<WebKit::DrawingAreaProxy, WebKit::DrawingAreaProxy, void (unsigned long, WebKit::LayerTreeContext const&), std::tuple<unsigned long, WebKit::LayerTreeContext> >(WebKit::DrawingAreaProxy*, void (WebKit::DrawingAreaProxy::*)(unsigned long, WebKit::LayerTreeContext const&), std::tuple<unsigned long, WebKit::LayerTreeContext>&&) () at ./Source/WebKit/Platform/IPC/HandleMessage.h:133
#9  IPC::handleMessage<Messages::DrawingAreaProxy::EnterAcceleratedCompositingMode, WebKit::DrawingAreaProxy, WebKit::DrawingAreaProxy, void (unsigned long, WebKit::LayerTreeContext const&)>(IPC::Connection&, IPC::Decoder&, WebKit::DrawingAreaProxy*, void (WebKit::DrawingAreaProxy::*)(unsigned long, WebKit::LayerTreeContext const&)) () at ./Source/WebKit/Platform/IPC/HandleMessage.h:235
#10 WebKit::DrawingAreaProxy::didReceiveMessage(IPC::Connection&, IPC::Decoder&) ()
    at DerivedSources/WebKit/DrawingAreaProxyMessageReceiver.cpp:48
#11 0x00007ffff26e75fa in IPC::MessageReceiverMap::dispatchMessage(IPC::Connection&, IPC::Decoder&) ()
    at ./Source/WebKit/Platform/IPC/MessageReceiverMap.cpp:129
#12 0x00007ffff27ed50a in WebKit::WebProcessProxy::didReceiveMessage(IPC::Connection&, IPC::Decoder&) ()
    at ./Source/WebKit/UIProcess/WebProcessProxy.cpp:1212
#13 0x00007ffff26e2195 in IPC::Connection::dispatchMessage(WTF::UniqueRef<IPC::Decoder>) ()
    at ./Source/WebKit/Platform/IPC/Connection.cpp:1451
#14 0x00007ffff26e2b3c in IPC::Connection::dispatchMessage(WTF::UniqueRef<IPC::Decoder>) ()
    at ./Source/WebKit/Platform/IPC/Connection.cpp:1408
#15 IPC::Connection::dispatchIncomingMessages() () at ./Source/WebKit/Platform/IPC/Connection.cpp:1563
#16 0x00007ffff1515995 in WTF::Function<void ()>::operator()() const () at ./Source/WTF/wtf/Function.h:82
#17 WTF::RunLoop::performWork() () at ./Source/WTF/wtf/RunLoop.cpp:147
#18 0x00007ffff15aa9e9 in operator() () at ./Source/WTF/wtf/glib/RunLoopGLib.cpp:80
#19 _FUN() () at ./Source/WTF/wtf/glib/RunLoopGLib.cpp:82
#20 0x00007ffff15ab6af in operator() () at ./Source/WTF/wtf/glib/RunLoopGLib.cpp:53
#21 _FUN() () at ./Source/WTF/wtf/glib/RunLoopGLib.cpp:56
#22 0x00007ffff6d8b67f in g_main_dispatch (context=0x55555ee87ec0) at ../../../glib/gmain.c:3454
#23 g_main_context_dispatch (context=context@entry=0x55555ee87ec0) at ../../../glib/gmain.c:4172
#24 0x00007ffff6d8ba38 in g_main_context_iterate
    (context=0x55555ee87ec0, block=block@entry=1, dispatch=dispatch@entry=1, self=<optimized out>) at ../../../glib/gmain.c:4248
#25 0x00007ffff6d8bcef in g_main_loop_run (loop=loop@entry=0x55555f3da310) at ../../../glib/gmain.c:4448
#26 0x00007ffff7408495 in gtk_main () at ../../../gtk/gtkmain.c:1329
#27 0x000055555af40d01 in wxGUIEventLoop::DoRun() (this=0x555560e96d00)
    at /OrcaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/gtk/evtloop.cpp:61
#28 0x000055555b055b11 in wxEventLoopBase::Run() (this=0x555560e96d00)
    at /OrcaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/evtloopcmn.cpp:87
#29 0x000055555b02b83a in wxAppConsoleBase::MainLoop() (this=0x55555ee91220)
[... cut ...]
==3569797==    by 0x1E4B2A6: wxAppConsoleBase::CallOnInit() (app.h:93)
==3569797==    by 0x5C33791: wxEntry(int&, wchar_t**) (init.cpp:488)
==3569797==    by 0x5C338AF: wxEntry(int&, char**) (init.cpp:516)
==3569797==    by 0x1DFDA1D: Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) (GUI_Init.cpp:64)
==3569797==    by 0x72C8D2: Slic3r::CLI::run(int, char**) (OrcaSlicer.cpp:1169)
==3569797==    by 0x77AB95: main (OrcaSlicer.cpp:6306)
==3569797== 
```

My version is v2.2.0 aka 1088d0 running on Debian 12 (Bookworm) on x86_64.

max-reynolds, 2025-02-05:
I observe the same issue 

>Thread 1 "orcaslicer_main" received signal SIGSEGV, Segmentation fault.
>webkitWebViewBaseEnterAcceleratedCompositingMode () at ./Source/WebKit/UIProcess/API/gtk/>WebKitWebViewBase.cpp:2938

This is actually a WebKitGtk problem as it can be reproduced with their test program MiniBrowser connecting to the Mainsail UI.

liberodark, 2025-02-15:
Hi,

For fix this issue have switch to flatpak.

Best Regards

TYTheBeast, 2025-02-15:
> switch to flatpak.

I hope you're tugging on my strings here, but, there is no such thing as OrcaSlicer on Flathub, nor is there a Flatpak release in the releases section. You're the first contributor to take notice of this thread, where multiple people are running into a **segmentation fault**, and your solution is to switch to the Flatpak?

First off, I'd like to ask, why the Flatpak? Is it the safest or most stable release? If so, why isn't the Wiki up to date to reference this?
Second, wouldn't it raise some concern that multiple people are getting a segmentation fault? This to me seems like a perfect vector for a specific individual attack, or to be nitpicky, a Memory Injection MITM SegFault Attack.
If this is not a possibility, then great, I'm worrying over nothing! 😅
But if this is a possibility, and if there's anything I can do to help resolve this, please do let me know! I am concerned about the security of OrcaSlicer, as it is built in C++, a really old language.

Have a great day 😄

TexZeTech, 2025-02-25:
What I find interesting is this doesn't seem to be an issue on X11 but does happen on Wayland.

I'm currently using the env variables `__GLX_VENDOR_LIBRARY_NAME=mesa __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json MESA_LOADER_DRIVER_OVERRIDE=zink GALLIUM_DRIVER=zink WEBKIT_DISABLE_DMABUF_RENDERER=1`

I don't think I need the env string on x11 but I'll verify that and edit this post.

> Operating System: Arch Linux 
> KDE Plasma Version: 6.3.1
> KDE Frameworks Version: 6.11.0
> Qt Version: 6.8.2
> Kernel Version: 6.12.16-1-lts (64-bit)
> Graphics Platform: Wayland
> Processors: 8 × Intel® Core™ i7-3770K CPU @ 3.50GHz
> Memory: 15.6 GiB of RAM
> Graphics Processor: NVIDIA GeForce GTX 1080

terrydervaux, 2025-03-08:
Managed to reproduce the bug on the following releases:

- `OrcaSlicer_Linux_AppImage_V2.3.0-beta2.AppImage`
- `OrcaSlicer-Linux-flatpak_V2.3.0-beta2.flatpak`
- `OrcaSlicer_Linux_AppImage_V2.3.0-beta.AppImage`
- `OrcaSlicer_Linux_V2.2.0.AppImage`

Environment:
```bash
$ uname -a
Linux xxxxx 6.8.0-52-generic #53~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Wed Jan 15 19:18:46 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
```

a0405u, 2025-05-15:
The issue is relevant for all the versions including AppImage, Flatpak, bin and building from source on Arch on both stable and dev versions.

tabiznet, 2025-06-05:
Same issue on v2.3.0 appimage only I can't even launch orca anymore it just crashes on start :(
Anyone know how to undo the device config without wiping my custom profiles?

RagingRoosevelt, 2025-08-02:
Also experiencing the issue on v2.3.0 flatpack on 22.04 PopOS.  

Confirming the workaround identified by @DiThi:  Put any text into the Device UI field which prevents the Mainsail / Fluid interface from loading.

Tenn0, 2025-10-04:
I have a workaround. My issue: AppImage and flatpak produced no rendering (thx nvidia). Got that fixed (dunno which command was it, sorry), but with klipper connection configured, it lead to a crash as soon as ive opened the device tab. Tried everything chatgpt was willing to give me. Nothing worked. Tried compiling orca from scratch natively, no luck building, since there was no build script for my distro CachyOS, i tried a few things but got tired of that. My solution: build it with docker and use docker to run it. 
Ill attach my DockerRun.sh script

[run.sh](https://github.com/user-attachments/files/22704441/run.sh)

Tenn0, 2025-10-04:
I have a workaround. My issue: AppImage and flatpak produced no rendering (thx nvidia). Got that fixed (dunno which command was it, sorry), but with klipper connection configured, it lead to a crash as soon as ive opened the device tab. Tried everything chatgpt was willing to give me. Nothing worked. Tried compiling orca from scratch natively, no luck building, since there was no build script for my distro CachyOS, i tried a few things but got tired of that. My solution: build it with docker and use docker to run it. 
Ill attach my DockerRun.sh script
note: you need xorg-xhost and nvidia-container-toolkit to be installed. `sudo pacman -S xorg-xhost nvidia-container-toolkit`

My setup is CachyOS with an RTX3070ti and KDE-Plasma. 
[run.sh](https://github.com/user-attachments/files/22704634/run.sh)

Karl-Barx, 2025-12-19:
> Also experiencing the issue on v2.3.0 flatpack on 22.04 PopOS.
> 
> Confirming the workaround identified by [@DiThi](https://github.com/DiThi): Put any text into the Device UI field which prevents the Mainsail / Fluid interface from loading.

Same issue on Pop for me. I had it working until the most recent Orca release