OrcaSlicer/OrcaSlicer #8244: Aligne Gaps doesn't work on small objects?

ForenSeil, 2025-01-29:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

i7 6th Gen
16Gig Ram
1TB SSD

### Printer

Bambu P1S

### How to reproduce

When slicing small objects Z-seams are not automatically aligned despite it's activated in the settings

Manual aligning does not work well because minimal stamp size is slightly too big for .2mm nozzle and very small objects

### Actual results

Cosmetically bad surface quality becaus Z-Seams are on random places everywhere

### Expected results

Gaps/Z-Seams should be automatically aligned.

### Project file & Debug log uploads

[Dummy+13+-+C3PO+Head_Sample_3MF.zip](https://github.com/user-attachments/files/18591996/Dummy%2B13%2B-%2BC3PO%2BHead_Sample_3MF.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

### Anything else?

[image]

discip, 2025-01-31:
@ForenSeil
You should check you per object setting, as you set Random for **`Seam position`** there. 😄

[image]

discip, 2025-01-31:
@ForenSeil
You should check you per object setting, as you set Random for **`Seam position`** there. 😄

[image]

Please don't forget to close this issue.

closed 2025-05-09 by github-actions[bot]