OrcaSlicer/OrcaSlicer #8250: Multimaterial ought to recognise when only layer switches are required and skip purge blocks

yoyo42, 2025-01-30:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

I'm just starting on my multimaterial journey but I've done a lot of 'sign mode' multicolour where you just swap the filament at a layer boundary. I can't see a way to use Orca for that style, either I just have to insert a pause and remember what is needed where or else I enable multimaterial and it prints a full height purge block which is a complete waste.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

It would be good if Orca could recognise that solid layers of colour are being used and skip the purge block completely, just swap the colours from the AMS at the appropriate times or even tell the user what to swap to manually if there is no AMS. 

### Additional context

'sign mode' is where a base plate is printed in one solid colour then text, logos etc are printed in another. The principle described above would work for that but also for models where colours go in complete layers and then return to the original colour.

lwbsbde, 2025-02-02:
der Reinigungsturm kann in den Einstellungen deaktiviert werden.
zusätzlich kann man auch der Reinigungsturm 

[image]

auch nur beim Filamentwechsel genutzt werden, und ist somit kleiner.

ShaneDelmore, 2025-02-05:
My guess is that the purge block is printed every layer to keep it the same height to avoid collisions, but it would be cool if orca could detect when the purge block is placed far enough away  to avoid collisions with the print head (similar to per object print order) and in those cases not waste the filament to keep it the same height.