OrcaSlicer/OrcaSlicer #8261: Save Presets for Individual Settings

labeshops, 2025-01-31:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

It's a pain to have to go thru settings like support and try to remember (or write down) what I used that worked well in one project for a different one. 

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Would be nice if you could save settings for sections like support, strength, etc. For example if I use the same tree support settings on multiple projects, instead of having to go thru each setting every time, would be nice if at the top of the Support tab I could select "save support settings" and have a drop down box to select those support presets again for a different project. 

### Describe alternatives you've considered

Yes, you can reload a project, clear the build plate and use that projects settings for a new plate, but this gets complicated and its easy to miss changing a setting you needed on the first project you do not need on this one. Being able to mix/match settings you know you like for things like support, strength, quality, etc. would be more elegant and useful.

Gugarutz, 2025-02-01:
would be an interesting option to have

hagre, 2025-02-01:
Perfekt for Multi Tool/ Material

kylek29, 2025-02-06:
This is something I also suggested on the PrusaSlicer GitHub (or here?) a long time back, I think it still has merit, and I'm glad someone else is bringing it back up.

The current settings overlap (inherit) Printer -> Filament -> Process -> Project (model, plates, etc.) , I believe .. with any user changes made in Project being the last. This would give the ability to save out the user-changes assigned as "Presets." Right now you have to fork Process settings for very minor things or remember what you need to change.

I'd imagine such a feature would need a UI modal to select what settings (or reuse the compare) to save into the preset, and some form of user-assignable QC-rules (e.g. not compatible with X-filament type, etc.), as well as the related override/load confirmation dialog.