OrcaSlicer/OrcaSlicer #8266: Flush into object not working correctly when 3 or more colors in same layer

Izew, 2025-01-31:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Bambu Lab P1P

### How to reproduce

Import 2 objects to the project. Paint one of them with 3 colors: dark green (#008000), white (#FFFFFF) and orange (#FF8000). Set the other of them as flush object (right click the object and select Flush options > Flush into this object. Slice the project and see the preview.

### Actual results

All the layers which have 3 different colors on them, are printed white on the flush object.

[image]

### Expected results

Flush object is used on every layer, also those that have 3 colors. Flush object looks stripey on the preview.

[image]

### Project file & Debug log uploads

[flush_object_does_not_work.zip](https://github.com/user-attachments/files/18623299/flush_object_does_not_work.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

### Anything else?

Flush into object is not working as intended when there's 3 or more colors on same layer. With two colors, it always works without issues. With 3 colors, most of the time flushing into object is ignored completely and the flush object is just printed in one solid color.

It only works correctly with certain colors, for example dark green (#008000), red (#ff0000) and white (#ffffff). It seems to be related to the flush volumes, and these colors happen to have the correct values. (In the attached project file, if the flush volume from orange to white is changed to 690, the flush object looks correct in the preview.) While upping the flush volume can be used as a way to circumvent the issue, it kinda defeats to purpose of using flush objects to minimize waste, as one has to flush much more than it actually is necessary to get the flush object to work.

[image]

[image]

fellu, 2025-02-01:
I have noticed this same behaviour o.O

Noisyfox, 2025-02-09:
I don't think that's a bug since the flush of colors between orange and green can be done within the wipe tower alone (due to the size of the tower). Only transaction between white filament requires a significant amount of purge hens the flush object is used.

Noisyfox, 2025-02-09:
It's working as expected. Note, stripping colors are not what this option is meant for. As long as the flush happens properly, it doesn't always produce the stripped pattern due to the ordering of the color.

Izew, 2025-02-09:
> It's working as expected. Note, stripping colors are not what this option is meant for. As long as the flush happens properly, it doesn't always produce the stripped pattern due to the ordering of the color.

It does not work as intended. There's no flushing happening into the object. Everything is flushed into poop chute, and then the flush object is printed in one solid color.

Noisyfox, 2025-02-09:
That's a different issue then. BambuLab printers always flush everything into pop bin (instead of the prime tower flush volume) I believe.

Izew, 2025-02-09:
> That's a different issue then. BambuLab printers always flush everything into pop bin (instead of the prime tower flush volume) I believe.

Yeah this is not at all what I was talking about. Sorry, I'm not native english speaker, so I struggle to explain this in a way it's easy to understand. But I'll try.

Here in the picture below you can see the flush volumes which are somehow above the magic limit after which the flushing into object works correctly. Let's look at the number from **orange to white**, which is 690. Then look at the table on the top of the picture. There you can see the actual flushed amounts, and these are the amounts that go into the poop chute (and not into the flush object). Total amount of flushed filament is 143,08 grams.

[image]

If I lower the flush volume of orange to white to 670, total amount of flushed filament should go down, right? However, it doesn't. It goes up. With lower flush volume the total amount of flushed filament is 159,70 grams, and that is because now none of the flushed filament goes into the flush object during the layers that have 3 different colors (which is the issue I'm trying to communicate here).

[image]

If there are only two colors per layer, total amount of flushed filament goes down as expected, as you can see in these two images below:

[image]

[image]

I hope this cleared a bit what I was trying to convey here.

closed 2025-05-18 by github-actions[bot]