OrcaSlicer/OrcaSlicer #8271: Installation file for Windows ARM64

Thomsen1982, 2025-02-01:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

no Windows ARM64 installation file is available

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

if the installation files could be extended by one for Windows ARM64, that would be perfect.
So that OrcaSlicer can run natively on the computer. This would make the calculation and so on much more performant on these computers.

As background: my Windows pc has given up the ghost. so I got myself the latest Surface. I found it very good with ARM, enormous computing power and low power consumption.

I'm sure many people would be grateful for an installation file.

PathKnower, 2025-02-20:
+1 here, however we have same one here: https://github.com/SoftFever/OrcaSlicer/issues/5783

Felix14-v2, 2026-09-29:
There are aarch64 builds in nightly releases, so support is implemented, I guess?

[image]

[image]