OrcaSlicer/OrcaSlicer #8275: Qidi X-Plus 4 profile missing timelapse code

SinisterMinister, 2025-02-01:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Win10

### Additional system information

N/A

### Printer

Qidi X-Plus 4

### How to reproduce

Print with timelapse enabled. Check for timelapse after print completes.

### Actual results

No timelapse created.

### Expected results

Timelapse created.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

Need to port the timelapse gcode from Qidi Studio. I'll submit a PR shortly.

cochcoder, 2025-03-13:
@SinisterMinister 
Is your issue fixed if you remove the current time lapse G-Code and replace the layer change G-Code to the one below? 

```
{if timelapse_type == 1} ; timelapse with wipe tower
G92 E0
G1 E-[retraction_length] F1800
G2 Z{layer_z + 0.4} I0.86 J0.86 P1 F20000 ; spiral lift a little
G1 Y304 F20000
G1 X95 F20000
G92 E0
M400
TIMELAPSE_TAKE_FRAME
G1 Y324 F5000
G1 E[retraction_length] F300
G1 X65 F5000
G1 Y290 F20000
{elsif timelapse_type == 0} ; timelapse without wipe tower
TIMELAPSE_TAKE_FRAME
{endif}
G92 E0
SET_PRINT_STATS_INFO CURRENT_LAYER={layer_num + 1}
```

Hobby-Student, 2025-06-17:
Same here, OrcaSlicer 2.3.0 with Qidi X-Plus 4. Timelapse is found via search, but the option itself is missing.

closed 2025-09-25 by github-actions[bot]