OrcaSlicer/OrcaSlicer #8276: Better default bridging

DusekM98, 2025-02-01:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Default bridging orientation does not seem to be a optimal solution. 
E.g.:

[image]

or my previous issue #6145.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I propose two solutions:

- Either make a parameter "max bridging distance", that would separate the bridged shape into "subshapes" if the default bridging exceeds the specified length and fill them accordingly. 

- Or an even better solution. Split any shape into its convex subshapes and fill them with the shortest bridging orientation.

[image]

igiannakas, 2025-02-01:
You can use the dual internal bridge feature here to cover both cases as they would be oriented 90 degrees to each other.

https://github.com/SoftFever/OrcaSlicer/pull/8263

DusekM98, 2025-02-01:
I checked your PR and even though it seems to solve some other important issues I believe that it's not the proper solution to this issue. Bridging in two layers by 90 degrees would help for sure, but why should the slicer bridge the first layer in the absurdly long orientation in the first place?

igiannakas, 2025-02-01:
Because it’s extraordinarily difficult to create segments dependant on geometry for now. It’s a neat idea but quite difficult to implement.

DusekM98, 2025-02-01:
Maybe I wasn't really clear in the description of this issue, but I'm not talking about internal bridging. This is "normal" bridging over the air. So there is no sparse infill to support it.

igiannakas, 2025-02-01:
Ah gotcha!

Yeap that’s something that could be better. Point re. Complexity still stands though, the auto calculation logic is pretty complex and will need someone with more skill than me 😀

henrivdr, 2025-02-04:
If we need someone with more skill than you @igiannakas , we are doomed 😉

DusekM98, 2025-02-06:
@lousassole I read your issue. You mentioned including solid infill patterns for bridging, but realistically, the only useful one could be the forms of arcs, just like in the arc overhangs "invention" (for the self-supporting feature). But it would slow the bridging process a lot for sure, and I can't imagine the quality of the arcs being any better than a straight pulled line. I can see how it could kinda solve this issue, but I don't think the negatives would be worth it. However, I could be wrong and I will be blown away once I try your solution in the future. 😉

john-parton, 2025-06-03:
>     * Or an even better solution. Split any shape into its convex subshapes and fill them with the shortest bridging orientation.
> 
> 
> [image]

Something that might not be immediately obvious is that it might not be sufficient to just split the shape into concave shapes. 

If you look at "Proposed 2", you have three regions: top region, right region, corner region. For optimal bridging, the corner region should probably be printed after both the top region and the right region. If you don't do that, you will end up having large parts which are un-anchored.