OrcaSlicer/OrcaSlicer #8277: Layer shift when slicing along a partial contour

KoeSource, 2025-02-01:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 11 64x

### Printer

Qidi X-Plus3

### How to reproduce

1. Open the STL (using 0.8mm Nozzle)
2. Slice and Zoom in.

[image]

[image]

### Actual results

I get a Layer Shift, see Screenshot.
The layer shift is only along a certain contour (blue line).
It does not affect the entire model.

When I open the model in PrusaSlicer, it is sliced correctly.

Nominal: 8.00mm
Top Layer: 7.95mm
Bottom Layer: 7.42mm

Can anyone Reproduce this?
Do i something wrong?

### Expected results

No layer Shift.

Expected result (Prusa-Slicer)
[image]

### Project file & Debug log uploads

[Project.zip](https://github.com/user-attachments/files/18630243/Project.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

buzzhuzz, 2025-02-02:
This is a side effect of XY hole compensation feature.

[image]

I would not recommend have it persistently enabled in your profiles.

KoeSource, 2025-02-02:
OK thank you very much.

closed 2025-02-02 by KoeSource