OrcaSlicer/OrcaSlicer #8281: Improve cooling of overhangs by maintaining slow speed after overhang ends

CarVac, 2025-02-02:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

I find that when printing Polymaker ASA at high speeds, on overhangs on only one side of a part, the corner where the printer begins laying down the overhang has little warping, while the corner where the overhang ends curls badly.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I'd like the print head to stay slow for a short distance after the overhang so that the exit corner spends enough time under part cooling airflow.

[image]

### Describe alternatives you've considered

Another alternative could be perhaps setting the acceleration away from that corner *very* low to make the speed transition gradual.

Inner/Outer/Inner may help too.

igiannakas, 2025-02-02:
Typically the part cooling duct is not focusing on a single "point", but more on an "area". hence what you're requesting is already happening to an extent. As the nozzle approaches the fan cools ahead of the area that will be printed and as it moves away continues to cool for a bit after.

If you want it to spend more time in sharp transitions, reducing acceleration is the way to go - it will also give the benefit of improved surface finish due to less input shaper corrections. And it will reduce curling a bit as the filament doesnt get pulled as much when accelerating slower.

CarVac, 2025-02-02:
> hence what you're requesting is already happening to an extent.

My point is, if the toolhead is going 50mm/s on the overhang and the duct covers a radius of 10mm, the filament gets 0.2 seconds of cooling.

But at the end of the overhang, the toolhead accelerates to 200mm/s and the filament only gets 0.05 seconds of cooling.

> If you want it to spend more time in sharp transitions, reducing acceleration is the way to go

Then I want high acceleration everywhere else for speed, and low acceleration for the line after the overhang.

Noisyfox, 2025-02-09:
> In this case, being able to set acceleration with something like "[% base accel] after angle change of 'X' over 'Y'mm or [% wall length]" would be the solution.

That's already possible with the extrusion rate smoothing option.