OrcaSlicer/OrcaSlicer #8288: Add "skip object" handling to GUI while printing, similar to Bambu Handy app

cryptonym64, 2025-02-02:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

With the transition to LAN only mode and no internet on Bambu devices, the Handy app will no longer be usable to skip failed objects.

It's always been a question as to why the mobile app has this function and it's not in the slicer.

### Which printers will be beneficial to this feature?

Others

### Describe the solution you'd like

I'd like to see a GUI based method to skip a failed object on a multi-object bed.  This is currently done in the Handy App which many bambu users will no longer have once we isolate our printers from the control freaks at Bambu.

I'm also using the Home Assistant app, and their issue wiki shows they are making headway on adding it to HA. I am hoping that the research there, and maybe some code would be usable for the ORCA Slicer team to add this long missing function into ORCA.

### Describe alternatives you've considered

Home Assistant would likely become the goto for mobile access to replace Handy app

But, I spend a lot of time at PC and so having it in the slicer would be optimal.

### Additional context

for reference see:    
https://github.com/greghesp/ha-bambulab/issues/866

Noisyfox, 2025-02-09:
Unfortunately we can't do that due to the lack of such capibility in bambulab's closed source network plugin. Unless someone is willing to make and maintain an opensourced version of that plugin that is binary compatiable with bbl's.

cryptonym64, 2025-02-09:
Why not just send the mqtt commands directly? It appears in looking at their discussion they pick up parts of the active file via ftp (3mf) and that has the coordinates and IDs. Then they send mqtt commands.  Isn't the dll just a wrapper to using mqtt and ftp, sprinkled with some logic to go cloud or local? With cloud becoming useless, maybe refactor the dll out and just go all mqtt and sftp?

Chevan94, 2025-02-11:
The recently released version of Bambu integration for Home Assistant has implemented an alpha version of this feature, and it's working on my A1. I guess it relies on direct mqtt commands and extracting object data from 3mf files, so it's definitely possible to do without the network plugin. 
It remains to be seen if this can be implemented to Orca Slicer in any way, shape or form.

yzheka, 2025-06-27:
[image]
Bambu Connect has this button but it is disabled. I am using the latest version of Orca. Also I was able to skip objects with bambu handy until went on LAN only mode