OrcaSlicer/OrcaSlicer #8292: Cannot use spiral vase

TeakKey7, 2025-02-02:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0-dev a29d535bc4

### Operating System (OS)

Linux

### OS Version

Arch Linux 6.6.72-1-lts

### Additional system information

Framework 16
CPU: AMD Ryzen 7 7040HS
Memory: 32GB 5600MHZ DDR6
GPU: Radeon 780M (integrated) - amdgpu driver
AUR: orca-slicer-git (runs X-wayland)
Display Manager: swaywm (Wayland)

### Printer

Ender 3 V2 BTT SKR 3 Mainsail

### How to reproduce

1. Go to Process > Other > Spiral Vase (ON)
2. Click yes
4. Slice

### Actual results

At some point (the same on each model) it will freeze, then eventually segfault. It seems to be stuck on the final layer on "Generating G-code: layer 300"

### Expected results

I should be getting a spiral sliced print

### Project file & Debug log uploads

Terminal:
/usr/include/c++/14.2.1/bits/stl_vector.h:1149: std::vector<_Tp, _Alloc>::const_reference std::vector<_Tp, _Alloc>::operator[](size_type) const [with _Tp = double; _Alloc = std::allocator<double>; const_reference = const double&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
[1]    21213 IOT instruction (core dumped)  orca-slicer

### Checklist of files to include

- [x] Log file
- [x] Project file

### Anything else?

Works fine on AUR orca-slicer-bin 2.2.0. I will stick to that for now. This is low priority for me but may have useful info for the devs. Treat this is debugging info.

TeakKey7, 2025-02-02:
[image]

TeakKey7, 2025-02-02:
[error.zip](https://github.com/user-attachments/files/18634322/error.zip)

Noisyfox, 2025-03-01:
Have you tried the latest nightly?

TeakKey7, 2025-03-02:
It is now working as of 2.3.0 Beta2 release. Thank you for checking!

closed 2025-03-02 by TeakKey7

cyberhuman, 2025-06-27:
I still have this issue when using https://aur.archlinux.org/packages/orca-slicer-git.

```
/usr/include/c++/15.1.1/bits/stl_vector.h:1282: std::vector<_Tp, _Alloc>::const_reference std::vector<_Tp, _Alloc>::operator[](size_type) const [with _Tp = double; _Alloc = std::allocator<double>; const_reference = const double&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
```

[vase-broken.zip](https://github.com/user-attachments/files/20949314/vase-broken.zip)

[image]