OrcaSlicer/OrcaSlicer #8295: Bambu P1S device sub tabs and camera not working

cryptonym64, 2025-02-03:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0-dev #8256

### Operating System (OS)

Windows

### OS Version

Win11 24h2 26.100.2894

### Additional system information

CPU i7-11370h
RAM 32GB (15gb free)
Video Iris XE

### Printer

Bambu P1S

### How to reproduce

Go to device tab.
Click play under camera
Get  Please check the network and try.....[2:-10060]
Status tab seems functioning itself, fans, temps, AMS, etc
Click MicroSD and get no sign of life, just square with circled arrow in background of page, no files
Click HMS and page is 100% empty, not sure what is there normally though.
Click Update and page is 100% empty, but I'd expect that maybe on LAN only mode

### Actual results

see above

### Expected results

The camera should feed a live stream, and the file list should be readable on printer

### Project file & Debug log uploads

no project used, clean start and it fails. This is the 8256 build file that fixed the LAN only connection issues.

[debug_Sun_Feb_02_18_10_44_24216.log.zip](https://github.com/user-attachments/files/18634851/debug_Sun_Feb_02_18_10_44_24216.log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

### Anything else?

I did notice the very first line in log appears to have a typo:
can not find function bambu_network_get_model_id_from_desgin_id

"design" is misspelled as "desgin" perhaps???

Noisyfox, 2025-02-03:
> Click MicroSD and get no sign of life, just square with circled arrow in background of page, no files
> Click HMS and page is 100% empty, not sure what is there normally though.
> Click Update and page is 100% empty, but I'd expect that maybe on LAN only mode

These are expected.

Does send print to printer work?

cryptonym64, 2025-02-03:
@Noisyfox 

Yes, print button pops up a box to map filaments then it sends and prints fine. 

The printer connection issue happened a few more times, but I've not had it since earlier today. I've now exited and restarted a dozen times and each time the cam has worked. No idea why though.  I had it fail at least 20 times straight, and now it works. 🤷‍♂️

I see that not having the SD Files is expected in LAN mode, but am very curious why that would be the case since it is open via SFTP user bblp@accesscode credentials?  Seems odd to not have that able to download. 

Don't the HMS errors just come in as status messages that could be dynamically looked up or even read from a database within Orca to give the human friendly info in Orca?

closed 2025-05-13 by github-actions[bot]