OrcaSlicer/OrcaSlicer #8296: Overhangs are generated instead of bridges on small width, large length features

NaisuMK2, 2025-02-03:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Bambu P1S

### How to reproduce

Make a part which requires a bridge, where the 2 part that need to be connected are rectangular in nature, about 1:4 proportions seem to work as in the example:

[image]

When the amount of perimeters is greater than the width (2mm width in this case), exclusively overhangs are generated:

[image]

### Actual results

As external perimeters fill up all of the free space inside of the feature, the floating region has little to none points to anchor to:

[image]

When the amount of external walls is set too quite a bit higher than the width of the feature especially, a large part of the floating region is left without points to anchor to:

[image]

This is especially bad when using Inner/Outer and Inner/Outer/Inner wall order, as the floating region will start printing from the center, essentially printing in mid air.

### Expected results

The floating region should be generated where the entire area is anchored to the part, for example just as when using one perimeter:

[image]

A simple work around would be to set the amount of perimeters on the layer where a bridge happens to one, and 100% sparse infill density so that the rest of the model's strength isn't compromised.

### Project file & Debug log uploads

[bridge_overhang_example.zip](https://github.com/user-attachments/files/18635500/bridge_overhang_example.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

hotellonely, 2025-02-03:
I can replicate your result by setting the wall count to 3,

[image]

...

[image]
It looks like that you can bypass the problem by setting the bridge counterbore holes to partial or full sacrificial layer.

It's a problem that I've found for quite a while, however there's no sign that it's gonna be fixed anytime soon.

hotellonely, 2025-02-03:
[image]
Also, for some reason, Orca tends to leave a gap between the walls and the infills here.

Anyway I feel like the current status of overhang/bridging related part is quite... messy. Lots of different functions interfering each other. 

Extra perimeter feature started from Prusa iirc, **it can cause lots of problems for bridging,** especially when you're bridging without supports. I don't know how Prusa implemented it but for sure it's a pretty poor design.

theboee, 2025-02-12:
Believe this may also be what's happening with mine #7197 issue

ampp33, 2025-02-27:
> It looks like that you can bypass the problem by setting the bridge counterbore holes to partial or full sacrificial layer.

THANK YOU, I knew there was some sort of secret sauce solution here or some trick to help get around this.  It's a very unexpected solution, but setting "Bridge counterbore holes" to "Partially Bridged" helped me go from unanchored internal bridge lines that were printing in mid-air

[image]

to properly anchored bridges

[image]

This issue still needs to be fixed, but this seems to be a good temporary work-around.

Benik3, 2025-03-14:
I just also found this, when I wanted to print bridge test.
Bridge walls are detected as overhangs. 

[image]

Also setting "Extra perimeters on overhangs" cause printing into the air as first:

[image]
with disabled:

[image]
Still there is very small overlap for the bridge line to the wall.

The counterbore holes can really help in this situation:

[image]

henrivdr, 2025-03-15:
I agree that the root cause need to be fixed. I also had to do the wall count workaround.

lexxxel, 2025-06-14:
Ping