OrcaSlicer/OrcaSlicer #8303: The position of the Cut tool plane is moved along the Z axis independently of the plane orientation (after a rotation).

Acelya-9028, 2025-02-03:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Bambu Lab P1S

### How to reproduce

1. Open a .3mf file on Orca Slicer
2. Use the Cut tool
3. Rotate it to 90
4. Change the Z Cut position value
5. TADA !!

### Actual results

**The plane position is moved along the Z axis independently of the plane orientation like the red arrow**

[image]

### Expected results

**The plane position is moved along the Z axis dependently of the plane orientation like the green arrow**

[image]

### Project file & Debug log uploads

[Test.zip](https://github.com/user-attachments/files/18642911/Test.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

Acelya-9028, 2025-02-05:
This doesn't really make sense because currently, once turned, this option becomes useless...

Noisyfox, 2025-03-01:
> This doesn't really make sense because currently, once turned, this option becomes useless...

You're right, this does seem odd. Not sure why this was designed like this, even Prusa and Bambu did this the same way. We copied the code from Prusa hens that's how it is.

Though this value can't be set based on the plan orientation because where 0 should be then?

closed 2025-06-07 by github-actions[bot]