OrcaSlicer/OrcaSlicer #8305: Non-Planar Infill

candre23, 2025-02-03:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

The extra steps required to impliment non-planar infill via gcode processing script are kind of a nusiance.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I see that brick layers is getting very close to being merged, and that's awesome.  Since that strength-enhancing technique is more or less implemented, perhaps it's time to look at another recent development that can substantially improve the strength of prints: non-planar infill.  The feature is already available via a gcode post-processing script.  Hopefully it could be integrated directly without too much fuss.

https://github.com/TengerTechnologies/NonPlanarInfill

candre23, 2025-02-03:
> Beside it looks cool, will the printed part be tougher?

That is the theory.  CNC Kitchen is supposedly working on a video with strength testing the method, so we'll have hard data soon.

Felix14-v2, 2025-02-06:
> will the printed part be tougher?

Hypothetically it should drastically improve the shear strength by increasing the layer area. The more Z-shift you have, the more area you will get.

mfraser, 2025-02-07:
> > will the printed part be tougher?
> 
> Hypothetically it should drastically improve the shear strength by increasing the layer area. The more Z-shift you have, the more area you will get.

My Tech Fun has just done a video on it https://www.youtube.com/watch?v=YuLNVoUtaes

candre23, 2025-02-07:
> My Tech Fun has just done a video on it https://www.youtube.com/watch?v=YuLNVoUtaes

This is for brick layers, which have been tested by a few people and are unambiguously stronger.  I don't know of any reliable testing of non-planar infill yet.  But as I said earlier, the person who came up with the gcode script contacted CNC kitchen and they are supposedly working on tests and a video.

theboee, 2025-02-12:
> Beside it looks cool, will the printed part be tougher?

It should increase shear strength since the layer will no longer be completely flat. I doubt it will help with layer adhesion though.