OrcaSlicer/OrcaSlicer #8311: Flatpak Tries to Access Windows Media Player

Prazgreen1, 2025-02-04:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

V2.3.0

### Operating System (OS)

Linux

### OS Version

Fedora 41

### Additional system information

             .',;::::;,'.                
         .';:cccccccccccc:;,.             ----------------
      .;cccccccccccccccccccccc;.          OS: Fedora Linux 41 (Workstation Edition) x86_64
    .:cccccccccccccccccccccccccc:.        Host: Swift SF314-512 (V1.06)
  .;ccccccccccccc;.:dddl:.;ccccccc;.      Kernel: Linux 6.11.4-301.fc41.x86_64
 .:ccccccccccccc;OWMKOOXMWd;ccccccc:.     Uptime: 47 mins
.:ccccccccccccc;KMMc;cc;xMMc;ccccccc:.    Packages: 2643 (rpm), 57 (flatpak)
,cccccccccccccc;MMM.;cc;;WW:;cccccccc,    Shell: bash 5.2.32
:cccccccccccccc;MMM.;cccccccccccccccc:    Display (AUODA91): 1920x1080 @ 60 Hz (as 1536x864) in 14" [Built-in]
:ccccccc;oxOOOo;MMM000k.;cccccccccccc:    DE: GNOME 47.3
cccccc;0MMKxdd:;MMMkddc.;cccccccccccc;    WM: Mutter (Wayland)
ccccc;XMO';cccc;MMM.;cccccccccccccccc'    WM Theme: Mint-Y-Dark-Aqua
ccccc;MMo;ccccc;MMW.;ccccccccccccccc;     Theme: Mint-Y-Dark-Aqua [GTK2/3/4]
ccccc;0MNc.ccc.xMMd;ccccccccccccccc;      Icons: Adwaita [GTK2/3/4]
cccccc;dNMWXXXWM0:;cccccccccccccc:,       Font: Noto Sans (9pt) [GTK2/3/4]
cccccccc;.:odl:.;cccccccccccccc:,.        Cursor: Adwaita (24px)
ccccccccccccccccccccccccccccc:'.          Terminal: cosmic-term
:ccccccccccccccccccccccc:;,..             CPU: 12th Gen Intel(R) Core(TM) i5-1240P (16) @ 4.40 GHz
 ':cccccccccccccccc::;,.                  GPU: Intel Iris Xe Graphics @ 1.30 GHz [Integrated]
                                          Memory: 5.88 GiB / 15.32 GiB (38%)
                                          Swap: 739.75 MiB / 8.00 GiB (9%)
                                          Disk (/): 45.89 GiB / 197.78 GiB (23%) - btrfs
                                          Disk (/run/media/prazgreen/Encrypted): 2.02 MiB / 28.65 GiB (0%) - ext4
                                          Disk (/run/media/prazgreen/General): 255.11 MiB / 28.67 GiB (1%) - ext4 [Ex]
                                          Local IP (wlp0s20f3): 192.168.0.25/24
                                          Battery (AP18C7M): 100% [AC Connected]
                                          Locale: en_US.UTF-8

yes, I just pasted my fastfetch script

### Printer

Bambulab A1 Mini

### How to reproduce

1. Install Flatpak
2. Try to access A1 Mini Camera from the devices tab

### Actual results

The slicer simply gives the error:
The player is malfunctioning. Please reinstall the media player.

### Expected results

I should just be able to view the camera feed

### Project file & Debug log uploads

cannot find any logs for the flatpak version

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Prazgreen1, 2025-02-04:
By the way, you should know that others seem to have this issue on r/orcaslicer, and I cannot use the Appimage as when I try to access the camera, the slicer crashes

ITankForCAD, 2025-02-05:
Yeah, seeing the same issue on Fedora 41 6.12.11. The appimage crashes and the flatpak throws the media player error.

Prazgreen1, 2025-02-05:
> Yeah, seeing the same issue on Fedora 41 6.12.11. The appimage crashes and the flatpak throws the media player error.

I just resorted to using a 3rd party patch to the Flatpak
https://github.com/anarsoul/io.github.softfever.OrcaSlicer
I saw there was an update 15 hours ago to the Flatpak, so I'll try the new version which might be a patch

Prazgreen1, 2025-02-05:
> I just resorted to using a 3rd party patch to the Flatpak https://github.com/anarsoul/io.github.softfever.OrcaSlicer I saw there was an update 15 hours ago to the Flatpak, so I'll try the new [...]

nvm this. I checked the pull requests, and this did not get added. It is still open and has not been merged.

Prazgreen1, 2025-02-19:
I realized this is a duplicate of another issue. Also, it is solved in a recent patch that was applied today. Thus, this is closed for two reasons.

closed 2025-02-19 by Prazgreen1