OrcaSlicer/OrcaSlicer #8313: Expose Line Spacing as configuration per line type

be-meyer, 2025-02-04:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

No

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

The ability to control line spacing independently from line width would enhance print quality control in OrcaSlicer. This feature would complement the existing line width settings and provide more precise control over extrusion behavior.

Current Limitation:
- OrcaSlicer only allows setting line width, which doesn't provide sufficient granular control over how lines are placed
- The spacing between lines affects print characteristics like overlap and adhesion between adjacent extrusions and can significantly improve surface quality

Proposed Solution:
1. Add a new setting for Line Spacing by Line Type that can be configured as either:
   - % of nozzle diameter
   - Fixed mm value

### Describe alternatives you've considered

To my knowledge, there is no alternative. I've transfered all configurations from a superslicer profile to orca but do not achieve a similar surface quality as with for example elis superslicer profiles. The only difference I can find is that it sets different line spacings for each of the line types.

### Additional context

#7098

ShaneDelmore, 2025-02-05:
Can you share some pictures and project files from orca and superslicer for comparison?

igiannakas, 2025-02-06:
It does for one perimeter type only - external perimeter -> precise wall feature sets spacing to no overlap between the two adjacent extrusions. Also for internal and external bridges (as of a recent PR I raised).

I have never encountered a use case where I'd need to change the spacing of the internal wall lines and solid infill, so cannot understand the need for additional settings that bring complexity. Can you give some use cases please?

igiannakas, 2025-02-06:
PS. the issue you've linked said "control the overlap of perimeters and top/bottom solid infill."
That exists today:

[image]

be-meyer, 2025-02-06:
Superslicer has very detailed configuration options for the extrusion width and the spacing. One of the most commonly used superslicer profiles for Voron is the [Ellis profile](https://github.com/AndrewEllis93/Ellis-SuperSlicer-Profiles) which very precisely sets width and spacings on each of the line types. It achieves a very high surface quality thus it's very popular as a startingpoint. 

[image]

While I can transfer nearly all other parameters from this superslicer, this is one I can't transfer and in my test prints the top surface finish are not equal and way more rough.  

_There might be more reasons but I'm not an absolute expert in slicer technology._

igiannakas, 2025-02-06:
Top surface quality is handled by the flow rate which is the recommended way from Ellis’ guide. You should be able to achieve excellent top surface finish by adjusting this value, plus setting infill type (internal and top and bottom) to monotonic lines and top surfaces line width to 0.4mm.

[image]
[image]
[image]
[image]

Could you post a picture of the superslicer print vs the orca print side by side so I can see the artefact you’re talking about?

In my opinion, there is close to zero reason to set these and can be “risky” if you don’t know what they do exactly, as they can cause gaps or over extrusion between lines. 

Suggest a read here as it’s a pretty comprehensive guide into line spacing and the underlying flow math.  https://manual.slic3r.org/advanced/flow-math

The outermost spacing has some reason to it as it can result in better external wall finish and hence has been implemented.

The top surface spacing you shouldn’t need to change. There are newer, settings to tweak in orca - don’t forget the Ellis profiles are quite a few years old by now.

For example, you can adjust the small area flow compensation to give you more consistent top surface extrusion between large and small areas, as the small areas tend to get over extruded.

Also monotonic lines for all infill types is highly advised as it gives a smoother finish vs monotonic.

Finally you can adjust the overlap between infill and walls for both top and bottom surfaces and solid infill to mitigate pinholes and reduce over extrusion on the top layers.

goldjunge91, 2025-02-07:
> Superslicer has very detailed configuration options for the extrusion width and the spacing. One of the most commonly used superslicer profiles for Voron is the [Ellis profile](https://github.com/AndrewEllis93/Ellis-SuperSlicer-Profiles) which very precisely sets width and spacings on each of the line types. It achieves a very high surface quality thus it's very popular as a startingpoint.
> 
> <img alt="Image" width="724" src="https://private-user-images.githubusercontent.com/4281635/410547157-93a26e92-5ec8-4212-beda-73a649e7f313.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mzg5NjUwMjMsIm5iZiI6MTczODk2NDcyMywicGF0aCI6Ii80MjgxNjM1LzQxMDU0NzE1Ny05M2EyNmU5Mi01ZWM4LTQyMTItYmVkYS03M2E2NDllN2YzMTMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDIwNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTAyMDdUMjE0NTIzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MDFlMDY2MTU2NzNlYjU1ZDEzYjEzZmU5ZTc3MGZlNGJiYWQ0ZDJlY2QyM2E1ZjU3ZmY3YmY3ZTBlMjAzNTRlOCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.bYlqd7rL5zQiWZhEfCDgEueDck96_VT4sxkkn5BnHd4">
> While I can transfer nearly all other parameters from this superslicer, this is one I can't transfer and in my test prints the top surface finish are not equal and way more rough.
> 
> _There might be more reasons but I'm not an absolute expert in slicer technology._

Interesting. How is the Part Strength?  it is better or worse ? 
And have you Slice the same file with both slicer and compared the Movement?

igiannakas, 2025-02-07:
Read up on the flow math here to see why this isn’t needed. Orca already overlaps the perimeter semi circles by 100% so there is no gap between them. Reducing overlap further would over extrude the part, and increasing it would produce weaker line adhesion.

The section that talks about spacing path:

https://manual.slic3r.org/advanced/flow-math

dbaarda, 2025-07-17:
I don't know if anybody has noticed this, but all the complicated math at https://manual.slic3r.org/advanced/flow-math ends with the statement;

>In the past, several values were tried for overlap_factor, but some users were still reporting too sparse paths. A value of 1 is currently being used to guarantee that the error (which is always present) is fully on the side of abundant extrusion rather than lacking material.

This use of 1 for the overlap_factor results in it all being identical to assuming the line cross-section is rectangular, which is the simple approximation used by most other slicers. The math is kind of interesting/useful if any other overlap_factor is used, but looking at the code it's hard coded everywhere, and all the equations used are simplified/transformed around that assumption. The code and documentation could be simplified even further with (zero impact on the resulting output) by just using the equivalent rectangular assumption (nearly) everywhere. This means line-width == line spacing.

I have a feeling that the unnecessary complications around line-width, line-spacing, and overlap have contributed to the confusion and possible complications around things like "Precise Wall". I'm still not yet sure if my line width settings are actually setting line spacing, or the wider line-width of the theoretical-but-not-actually-used bulging lines.

Which does raise the one area where the non-rectangular assumption could possibly be useful; external perimeter walls. Here the circular bulge does mean it protrudes beyond the rectangular-model edge, which can matter for tight fitting parts. The current line-width/overlap-factor/etc approach effectively accounts for this, even without "Precise Wall", because there is no overlap-compensation for the outer-wall edge so it includes the outer "bulge" in the line width. However the rectangular approximation still gives you the accurate **average** of the perimeter wall, which is useful/important for interference fitting parts where the bulges tend to get squished flat, or when some kind of surface-smoothing post-processing is applied. This means the current approach effectively under-dimensions things for some applications. Note the difference between the the "bulges squished totally flat" edge and the "only just touches the circular bulge" edge is (1-pi/4)/2 times the layer height, which is only 0.02mm for a 0.2mm layer height. Such a tiny change in the outer dimensioning probably makes it not worth the added complexity of compensating for it, especially when you might not even want to compensate for it in some cases. 

What "Precise Wall" effectively does is slightly increase the outer-wall line-spacing without a corresponding increase in extrusion rate. This is the rectangular-model equivalent of a wider line with a reduced extrusion ratio. This can improve dimensioning accuracy if the rest of the print, and in particular the inner wall, is generally over-extruded, which is often done for strength. I always try to avoid too much over-extrusion because it messes with dimensioning too much. Precise Wall is also way too off/on, and in practice the degree of under-extrusion you want to compensate for any inner-wall over-extrusion is delicate and typically much less than what Precise Wall gives you.

All this explains why on switching to orca from (the admittedly crapy) FlashPrint I'm finding it consistently under-dimensions my designs, even with Precise Wall off.

TLDR; The current complicated flow math is equivalent to a simple rectangular model because it always uses overlap_factor=1. This means the distinction between line-width and line-spacing is an unnecessary complication better solved by just using line-width==line-spacing. Adjusting extrusion ratios can be used instead if different degrees of "overlap" or strength-vs-accuracy are desired.

dbaarda, 2025-07-29:
For people who have well tweaked/calibrated their systems coming to Orca from slicers that use the rectangular model where line-width=line-spacing, you can convert your old settings to match what Orca uses. I've checked Orka output and its line width settings are the theoretical rounded-bead width, so they are wider than the line_spacing. Convert the line-width or line-spacing of other slicers to Orca with the following;

Orca effectively uses thinner lines than other slicers by `layer_height * (1 - PI/4) = layer_height*0.2146`, or roughly layer_height/5. So add that much to your line_width setting. Note that the adjustment depends on layer_height, so if you have a different first_layer_height, have enabled Precise_Z_height (which can adjust layer heights to give a more accurate model height), or enabled infill combination (which can merge infill layers into one thick layer), then the adjustment will be off for those layers. Note that this also means Orca effectively uses narrower lines for thicker layers with the same line-width settings, which is IMHO the opposite of what you would want. For fine nozzle-width=0.4mm and layer_heights=0.1, 0.2, 0.3  I set orca line-widths as a % of the nozzle width and add 5%, 11% and 16% respectively.

Orca effectively shifts the outer-wall inwards by `layer_height * (1-PI/4)/2 = layer_height * 0.1073`, or roughly layer_height/10. This can be compensated for by adding it to "Quality; Precision; X-Y contour compensation" and subtracting it from "Quality; Precision; X-Y hole compensation". The only thing I'm not entirely sure about is if the "X-Y contour compensation" and "X-Y hole compensation" distance settings are edge-to-edge (diameters), or offsets (radius). I know FlashPrint uses diameters, and if it's the same in Orca then the wall offset adjustments need to be doubled for those settings. 

Note "Precise Wall" doesn't help here, and makes it worse by making the outer wall wider and under-extruded, so I'd turn that off. As mentioned above the outer-wall shifting by Orca is accounting for the "layer bulge" effect so the outer edge of the layer bulge matches the model, whereas other slicers have the "average edge" matching the model. So applying the outer-wall adjustment makes it match your settings on other slicers, but if you want a non-interference fit you might not want to include it.

Note Orca has many other settings like "Convert holes to polyholes", and "Arc fitting" that can affect these, but I'm not yet sure how useful they are and if/how to use them. I'd turn them off unless you want to go through re-calibrating everything with them.