OrcaSlicer/OrcaSlicer #8317: Error when trying to connect to Prusa connect

Beukie66, 2025-02-04:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0-dev build 0c19086

### Operating System (OS)

Windows

### OS Version

windows 10 latest update

### Additional system information

OS Name	Microsoft Windows 10 Pro
Version	10.0.19045 Build 19045
Other OS Description 	Not Available
OS Manufacturer	Microsoft Corporation
System Name	DESKTOP
System Manufacturer	HP
System Model	HP Z2 Tower G4 Workstation
System Type	x64-based PC
System SKU	2YW27AV
Processor	Intel(R) Core(TM) i7-8700K CPU @ 3.70GHz, 3696 Mhz, 6 Core(s), 12 Logical Processor(s)
BIOS Version/Date	HP Q50 Ver. 01.08.08, 12-10-2023
SMBIOS Version	3.1
Embedded Controller Version	7.210
BIOS Mode	UEFI
BaseBoard Manufacturer	HP
BaseBoard Product	8455
BaseBoard Version	KBC Version 07.D2.00
Platform Role	Desktop
Secure Boot State	On
PCR7 Configuration	Elevation Required to View
Windows Directory	C:\WINDOWS
System Directory	C:\WINDOWS\system32
Boot Device	\Device\HarddiskVolume3
Locale	Netherlands
Hardware Abstraction Layer	Version = "10.0.19041.5072"
User Name	DESKTOP
Time Zone	W. Europe Standard Time
Installed Physical Memory (RAM)	64,0 GB
Total Physical Memory	63,8 GB
Available Physical Memory	50,8 GB
Total Virtual Memory	192 GB
Available Virtual Memory	176 GB
Page File Space	128 GB
Page File	C:\pagefile.sys
Kernel DMA Protection	Off
Virtualization-based security	Not enabled
Device Encryption Support	Elevation Required to View
Hyper-V - VM Monitor Mode Extensions	Yes
Hyper-V - Second Level Address Translation Extensions	Yes
Hyper-V - Virtualization Enabled in Firmware	Yes
Hyper-V - Data Execution Protection	Yes

### Printer

Prusa XL5T and Prusa MK4 Is

### How to reproduce

After setting connection to Prusa connect. Test OK!

Goto Tab Device

Login page prusa connect is visible

When doing login error message failed to fetch

see attachment

[image]

### Actual results

[image]

### Expected results

connection to web page prusa connect

### Project file & Debug log uploads

[TEST Cube.zip](https://github.com/user-attachments/files/18659508/TEST.Cube.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ericus256, 2025-02-05:
I have also been experiencing this issue, essentially verbatim to @Beukie66's description and screenshots. I started seeing it at least a month ago, but figured it was some server issue at Prusa (and sending actual print jobs seemed to work even though the Device tab failed out like that). But alas, it has not resolved itself in all this time, so chiming in here.

In Prusa Slicer, logging in with Prusa account in the top-right results in Prusa Connect tab appearing and automatically cutting-through to Prusa Connect dashboard without an additional login and 2FA prompt. So Prusa Connect itself seems fine but something changed in how Orca Slicer is trying to load the page(s) and/or communicate with Prusa Connect.

jan1409, 2025-02-09:
Same problem here. Exactly how it is described above. Since a couple of weeks I cannot use Prusa Connect anymore and getting these errors. I have no problem connecting via Browser (tried Firefox and Edge) and also not in the Prusa slicer itself. 
The page is just asking me to connect (Logon) and then the issue occurs. Are there some certificates or other stuff stored in Orca slicer which might contain "outdated" or expired logon credentials? (e.g. cookies)
The reason does not seem to be with the Orca slicer version since I know that it worked with the same version after I installed it (latest & Greatest). 
Would

1activegeek, 2025-02-17:
Will chime in here as well I'm having this issue. It does seem like something that was not an issue earlier (though I haven't printed for a bit with Orcaslicer to my Prusa). I will add however, that sending of files seems to still work perfectly fine. So I can in fact send prints to the printer (thankfully!) but I can't monitor what's going on unless I open up PS or the browser and navigate to the Connect page. 
On macOS Sequoia 15.3 using OrcaSlicer 2.2.0 and 2.3.0-dev, same results.

donalffons, 2025-03-04:
Just trying out OrcaSlicer for the first time and I'm also running into this issue. By enabling "Developer Mode" in the preferences, it is possible to get more detailed error messages from Gecko (?):

[image]

This looks like a CORS / CSP issue, perhaps?!

donalffons, 2025-03-04:
Or maybe not... When I right-click the problematic request to `account.prusa3d.com/o/token/` in the "Network"-Tab and choose "Copy as cURL", then execute that curl command in a terminal, the prusa server responds with an error, which doesn't include any CORS / CSP headers - but that might not be the root cause of the issue:
```
$ curl -o - 'https://account.prusa3d.com/o/token/' \
-X 'POST' \
-H 'Content-Type: application/x-www-form-urlencoded' \
-H 'Origin: https://connect.prusa3d.com' \
-H 'Referer: https://connect.prusa3d.com/' \
-H 'Accept: */*' \
-H 'Sec-Fetch-Dest: empty' \
-H 'Sec-Fetch-Mode: cors' \
-H 'Sec-Fetch-Site: same-site' \
-H 'User-Agent: BBL-Slicer/v01.10.01.50 (light) Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/605.1.15 (KHTML, like Gecko)' \
-H 'X-API-Key: blablabla' \
--data 'grant_type=authorization_code&code=blablabla&client_id=blablabla&redirect_uri=https%3A%2F%2Fconnect.prusa3d.com%2Flogin%2Fauth-callback&code_verifier=blablabla'
> {"error": "invalid_grant"}% 
```

Lungenpest, 2025-03-14:
Same here.

With Flatpak and AppImage on Manjaro.

PrusaSlicer connects flawless to Prusa Connect without any login problems.
But when i try to login in OrcaSlicer same error appears like TO.

Phardbean, 2025-03-18:
Having this as well. Everything prints.  Windows 11 blah blah. seems to be Orca/prusa specific?

Unexpected Application Error!
Failed to fetch
TypeError: Failed to fetch
    at window.fetch (<anonymous>:8:20)
    at Object.completeLogin (https://connect.prusa3d.com/assets/index-d53e2c42.js:1199:11612)
    at Nwt (https://connect.prusa3d.com/assets/index-d53e2c42.js:1199:13990)
    at f (https://connect.prusa3d.com/assets/index-d53e2c42.js:102:3306)
    at H0 (https://connect.prusa3d.com/assets/index-d53e2c42.js:102:3737)
    at https://connect.prusa3d.com/assets/index-d53e2c42.js:101:23462
    at Array.map (<anonymous>)
    at Oe (https://connect.prusa3d.com/assets/index-d53e2c42.js:101:23454)
    at oe (https://connect.prusa3d.com/assets/index-d53e2c42.js:101:18326)
    at U (https://connect.prusa3d.com/assets/index-d53e2c42.js:101:16495)

mr-potato-head, 2025-03-20:
Same here.

Orca v2.3.0-rc
Windows 11

prisoner06, 2025-03-21:
Same here 

Orca c2.3.0 Stable
Windows 11

bwees, 2025-03-23:
I did some investigation into the issue this evening. The issue appears to be caused by Prusa's servers not giving a correct response back when WebKit tries to do the CORS checking. The CORS errors we are seeing are a result of the browser enforcing what headers are being sent to the server. From what I can tell, PrusaConnect is not giving a valid CORS response when WebKit sends a `OPTION` request to the server. If the CORS request did not give valid data back, it basically locks down the request as much as possible and thus no headers are allowed to be sent to the server. 

This explains why @donalffons was able to get the request to go through via `curl` because curl does not do any CORS enforcement since it is not a browser.

I have sent prusa support a message explaining the issue and linking to this GitHub issue. If you have a spare moment, please open a [live chat](https://help.prusa3d.com/article/customer-support_2287) with them and report this issue. With enough of us doing this, it should bring enough awareness to the issue.

closed 2025-06-29 by github-actions[bot]

SkeletalPrints, 2025-08-27:
[image]
I am still getting something that looks about the same. Not sure if anything is different because it now a few months later

EEJoshua, 2025-09-28:
2.3.1-beta, same issue

[image]

EDIDDD, 2026-02-24:
Still having the same issue in 2026 :
Unexpected Application Error!
Failed to fetch
TypeError: Failed to fetch
    at window.fetch (<anonymous>:8:20)
    at Object.completeLogin (https://connect.prusa3d.com/assets/index-Dm6kNaKg.js:1410:11133)
    at akt (https://connect.prusa3d.com/assets/index-Dm6kNaKg.js:1410:14065)
    at h (https://connect.prusa3d.com/assets/index-Dm6kNaKg.js:653:5031)
    at https://connect.prusa3d.com/assets/index-Dm6kNaKg.js:653:5155
    at l (https://connect.prusa3d.com/assets/index-Dm6kNaKg.js:653:5202)
    at tYe (https://connect.prusa3d.com/assets/index-Dm6kNaKg.js:653:5574)
    at Object.resolve (https://connect.prusa3d.com/assets/index-Dm6kNaKg.js:653:4560)
    at https://connect.prusa3d.com/assets/index-Dm6kNaKg.js:653:4224
    at Array.map (<anonymous>)

thijs-philips, 2026-10-25:
`TypeError: Failed to fetch
    at window.fetch (<anonymous>:8:20)
    at Object.completeLogin (https://connect.prusa3d.com/assets/jwt-d1lRrL00.js:1:4231)
    at H (https://connect.prusa3d.com/assets/jwt-d1lRrL00.js:1:7602)
    at u (https://connect.prusa3d.com/assets/chunk-KS7C4IRE-DqLwmI_O.js:1081:9984)
    at https://connect.prusa3d.com/assets/chunk-KS7C4IRE-DqLwmI_O.js:1081:10251
    at f (https://connect.prusa3d.com/assets/chunk-KS7C4IRE-DqLwmI_O.js:1081:10298)
    at Is (https://connect.prusa3d.com/assets/chunk-KS7C4IRE-DqLwmI_O.js:1081:10738)
    at Object.resolve (https://connect.prusa3d.com/assets/chunk-KS7C4IRE-DqLwmI_O.js:1081:8446)
    at https://connect.prusa3d.com/assets/chunk-KS7C4IRE-DqLwmI_O.js:1081:6615
    at Array.map (<anonymous>)`

me25542, 2026-11-08:
Same issue with 2.5.0 stable.

[image]